plerup/espsoftwareserial #136: readBytes timeout is only blocking for one character length rather than full read quantity

SRGDamia1, 2020-01-24:
I'm not a regular user or follower of the ESP core, so I don't know if this is intended behavior.

Expected:  readBytes(buf, 256) blocks for 256*_timeout ms.
Current:  readBytes(buf, 256) blocks for _timeout ms.

I maintain a modbus library and it was brought to my attention that it wouldn't work on the current ESP core's version of ESPSoftwareSerial (esp core 2.6.3 which has ESPSoftwareSerial 6.4.0), but did work on what version had been packaged with esp core 2.5.2.  It's because of the new implementation of readBytes which was non-blocking for the timeout, when my library was expecting blocking in the way an AVR operates.  I see that you've already changed it to be blocking with 6.5.0, but it's still not blocking for long enough.  The current behavior has a total block of only 1 timeout period to fill the entire buffer asked for with read bytes.  To line up with what AVR style Arduino's do (if that's the goal) it should block for 1 timeout length for each individual character being read out.  So if I'm reading into a long buffer, it should wait a while.  With slow sensors talking at 9600baud, the timeout is reached while there are still characters that should be coming.

dok-net, 2020-01-25:
@SRGDamia1 Thanks for the observation! With a slight disagreement on the conclusion regarding overall timeout, I've fixed the implementation in EspSoftwareSerial 6.7.1 now. In short, we don't block on `size` * * `_timeout_ms`, which IMHO can turn out pretty badly, but just like the Stream class in ESP8266 we block up to `_timeout_ms` for each individual delay. So now, if data is coming in slowly but steadily, readBytes() does not return prematurely, but neither does it unconditionally block for a very long amount of time if there is a large buffer, but no data at all.

dok-net, 2020-01-25:
@plerup Hi Peter, I've permitted myself to change the maintainer email address to point at me, I think it's just and makes the most sense in our current arrangement. If this doesn't please you, just let me know and I'll reverse that.

closed 2020-01-25 by dok-net

SRGDamia1, 2020-01-25:
Thank you for the fast change!

commit 9a30d839 (2020-01-25): Fix #136.

Replicate behavior of ESP8266 Stream class, may possibly diverge from readBytes() on AVR:
Timeout must be considered for any period of not receiving any data. Before, it applied unconditionally
to the readBytes() call overall.
Considering body of work over the past year, changing maintainer entry in library descrptor to myself.