plerup/espsoftwareserial #138: Rx line getting messed up

viktak, 2020-02-07:
Hi,
I have been using an old version of this library for years with no major problem. Today I decided to update my projects to the latest version of this library and ran into a blocking issue:
Once the espsoftwareserial object gets started, it messes up the serial line (Rx), as a result no readings can be made. With the oscilloscope I figured this out:
Up until version 3.4.1 the signal on the Rx pin looks like [this](http://www.viktak.com/test/serial_normal.bmp), with any version after that (I tried a few at random) it looks like [this](http://www.viktak.com/test/serial_messedup.bmp).
This is the offending code:
`swSer.begin(9600, SoftwareSerialConfig::SWSERIAL_8N1, SOFTWARESERIAL_RECEIVE_GPIO, SOFTWARESERIAL_SEND_GPIO, false, 640, 0);`
Any idea what it could be?
Thanks for any help in advance!
viktak

dok-net, 2020-02-08:
I turned on PULLUP on the GPIO for RX at two times during past development, I have to let the comments on the commits speak for themselves, because it's been a while since then:
`Turn on internal pullups on RX pin. Improves SNR significantly in ESP8266 nullmodem test.` (a3bda42377c0ac3b279f6cff95f7957aeb5f116a)
`ESP32 requires GPIO input pullup.` (db808f0d9bfab26e0e9f86ef76d6608fe16c48d5)
I would appear that your HW setup has to be adapted to the internal pullup on the RX GPIO.
Given successful testing between ESP32, ESP8266, internal HW to SW serial, as well as to the USB UARTs, with no issue at all using INPUT_PULLUP on the RX pin, the current mode is correct and will not be changed.
Please report, though, how you have wired up the RX/TX GPIOs during in-circuit test, or how your oscilloscope measurement were produced.

closed 2020-02-08 by dok-net

viktak, 2020-02-09:
It was indeed the issue.
What i did to circumvent it (temporarily) is I added an extra line of code to disable the internal pullups:
`  swSer.begin(9600, SoftwareSerialConfig::SWSERIAL_8N1, SOFTWARESERIAL_RECEIVE_GPIO, SOFTWARESERIAL_SEND_GPIO, false, 640, 0);
  pinMode(SOFTWARESERIAL_RECEIVE_GPIO, INPUT);
`
At this stage I'm pretty sure I won't change the existing hardware, maybe a future revision.The current one is nothing special: I connect the Tx line of a GPS receiver directly to the Rx line of the ESP8266. With my extra line of code above (disabling the pull-up) it works perfectly.

The mechanical setup is that the GPS receiver is outside the house, the ESP is on my desk and the cable in between is around 10 meters.

The oscilloscope is attached directly to the cable running from the GPS to the ESP.

What hardware modification would you suggest? I would like to avoid adding extra components to the board, if possible.