plerup/espsoftwareserial #141: Additional implicit copy constructor issues w/ gcc 9.2

earlephilhower, 2020-02-28:
I pulled in your master with the copy constructor fix to the GCC 9 PR for the 8266.  Looks like a new implicit copy constructor issue cropped up:

````
/home/travis/arduino_ide/hardware/esp8266com/esp8266/libraries/SoftwareSerial/src/SoftwareSerial.cpp: In member function 'void SoftwareSerial::onReceive(Delegate<void(int), void*>)':

/home/travis/arduino_ide/hardware/esp8266com/esp8266/libraries/SoftwareSerial/src/SoftwareSerial.cpp:547:22: error: implicitly-declared 'Delegate<void(int), void*>& Delegate<void(int), void*>::operator=(const Delegate<void(int), void*>&)' is deprecated [-Werror=deprecated-copy]

  547 |     receiveHandler = handler;

      |                      ^~~~~~~

In file included from /home/travis/arduino_ide/hardware/esp8266com/esp8266/libraries/SoftwareSerial/src/circular_queue/circular_queue.h:31,

                 from /home/travis/arduino_ide/hardware/esp8266com/esp8266/libraries/SoftwareSerial/src/SoftwareSerial.h:27,

                 from /home/travis/arduino_ide/hardware/esp8266com/esp8266/libraries/SoftwareSerial/src/SoftwareSerial.cpp:23:

/home/travis/arduino_ide/hardware/esp8266com/esp8266/libraries/SoftwareSerial/src/circular_queue/Delegate.h:149:28: note: because 'Delegate<void(int), void*>' has user-provided 'delegate::detail::DelegatePImpl<A, R, P>& delegate::detail::DelegatePImpl<A, R, P>::operator=(const delegate::detail::DelegatePImpl<A, R, P>&) [with A = void*; R = void; P = {int}]'

  149 |             DelegatePImpl& operator=(const DelegatePImpl<A, R, P...>& del)

      |                            ^~~~~~~~

/home/travis/arduino_ide/hardware/esp8266com/esp8266/libraries/SoftwareSerial/src/circular_queue/Delegate.h: In member function 'Delegate<void(int), void*>& Delegate<void(int), void*>::operator=(const Delegate<void(int), void*>&)':

/home/travis/arduino_ide/hardware/esp8266com/esp8266/libraries/SoftwareSerial/src/circular_queue/Delegate.h:1788:55: error: implicitly-declared 'delegate::detail::Delegate<void, void*, int>& delegate::detail::Delegate<void, void*, int>::operator=(const delegate::detail::Delegate<void, void*, int>&)' is deprecated [-Werror=deprecated-copy]

 1788 | template<typename R, typename A, typename... P> class Delegate<R(P...), A> : public delegate::detail::Delegate<R, A, P...>

      |                                                       ^~~~~~~~~~~~~~~~~~~~

/home/travis/arduino_ide/hardware/esp8266com/esp8266/libraries/SoftwareSerial/src/circular_queue/Delegate.h:149:28: note: because 'delegate::detail::Delegate<void, void*, int>' has user-provided 'delegate::detail::DelegatePImpl<A, R, P>& delegate::detail::DelegatePImpl<A, R, P>::operator=(const delegate::detail::DelegatePImpl<A, R, P>&) [with A = void*; R = void; P = {int}]'

  149 |             DelegatePImpl& operator=(const DelegatePImpl<A, R, P...>& del)

      |                            ^~~~~~~~

/home/travis/arduino_ide/hardware/esp8266com/esp8266/libraries/SoftwareSerial/src/SoftwareSerial.cpp: In member function 'void SoftwareSerial::onReceive(Delegate<void(int), void*>)':

/home/travis/arduino_ide/hardware/esp8266com/esp8266/libraries/SoftwareSerial/src/SoftwareSerial.cpp:547:22: note: synthesized method 'Delegate<void(int), void*>& Delegate<void(int), void*>::operator=(const Delegate<void(int), void*>&)' first required here

  547 |     receiveHandler = handler;

      |                      ^~~~~~~

cc1plus: all warnings being treated as errors
````

dok-net, 2020-02-28:
Maybe you can retry, it wasn't done just yet, now it should work.Sent from my Huawei Mobile

earlephilhower, 2020-02-29:
Looks good now.  Thx!

closed 2020-02-29 by earlephilhower