plerup/espsoftwareserial #142: espsoftwareserial with ESP8285.

homefefeld, 2020-03-07:
Hi guys,
the ESP8285 has two more GPIOs : CPIO9 and GPIO10.
I would like to use a Sonoff 2 channels, it has those pin evailable:
[image]
But it seems that I cant use it.  is that possible to add them?
Ideally , I would like to use 10,14.

Thank you.

G.

closed 2020-03-07 by dok-net

dok-net, 2020-03-07:
@homefefeld Here's a friendly warning, if you haven't been there before: EspSoftwareSerial is released as part of ESP8266 Arduino core, due to the handling as git submodule there. There are detailed instructions in the READMEs of both projects etc. on how to setup git clones and synchronize to the respective master or release points.

homefefeld, 2020-03-10:
Thank you. I'll try your fix.

commit d4b3c77b (2020-03-07): Bump release for ESP8285 support. Fixes #142