plerup/espsoftwareserial #143: adding an really easy to understand example including a picture of two ESP8266-boards  wired together

StefanL38, 2020-03-15:
Hello,

first off all thank you very much for providing an adapted softwareserial for ESP8266.
I appreciate that. 

I have read all the examples but I'm still very unsecure how to use it. The comments are too short as that I would understand them. 
There are a lot of ifdef-statements in the sourcecode. And almost nothing what looks like
a real testcode. So I would appreciate it very much if all the ifdef-stuff could be hided away in the includefile. And if there would be a really **easy to understand** exmaple be provided.

dok-net, 2020-03-16:
Alright, let's see.Servotester is obviously quite specific.Loopback and repeater are for regression and load/performance testing, leave them aside.Swsertest should attract those looking for a sample, onewiretest the same. If you cannot figure out those two, still it"s just a serial API that's Arduino HW and SW compatible, so look around everywhere else where there is such a thing.In short, EspSoftwareSerial is a drop-in replacement for HW serial, (almost) no assembly required. That said, I wouldn't mind receiving contributions for a canonical example 🙂

dok-net, 2020-03-19:
This has now essentially been duplicated in #144.

closed 2020-03-19 by dok-net