plerup/espsoftwareserial #145: ESP32 ModbusMaster + Softwareserial

GO74, 2020-03-19:
Hi, 
I'm working on a Modbus project. We had some issues with the hardwareserial and I wanted to swap in softwareserial.
The ESP32 (Olimex) crashes the moment Modbusmaster is trying to use it.
Any idea where to start digging?

dok-net, 2020-03-19:
Without any diagnostics in the minimum? Nope... Please supply MCVE, log output, version numbers. If noone else can run it, though, you're on your own.-------- Original Message --------Subject: [plerup/espsoftwareserial] ESP32 ModbusMaster + Softwareserial  (#145)From: GO74 To: plerup/espsoftwareserial CC: Subscribed 
Hi,
I'm working on a Modbus project. We had some issues with the hardwareserial and I wanted to swap in softwareserial.
The ESP32 (Olimex) crashes the moment Modbusmaster is trying to use it.
Any idea where to start digging?

dok-net, 2020-04-10:
No feedback given, closing.

closed 2020-04-10 by dok-net