plerup/espsoftwareserial #146: One wire tx / rx switching problem

zs6buj, 2020-03-22:
Nice work on this project. Thank you.

For ESP32 and ESP8266 boards I'm trying to use your 1-wire feature to eliminate the need for an external inverter/1-wire converter in [this project](https://github.com/zs6buj/MavlinkToPassthru/wiki). The project sends and receives half-duplex 1-wire telemetry between the ESP and FrSky brand rc receivers (or transmitter boxes) using their "S.Port".  We are doing this successfully in hardware using a Teensy 3.x.

I have looked at and tested your OneWire example, and , can successfully send and receive on 1-wire inverted using my own small test module, but the switching appears to fail when I use it in my much larger  project. I suspect there might be contention for resources between WiFi and ESPSWSerial.

This is the pertinent code:

First initialisation:

```
  #if ((defined ESP8266) || (defined ESP32) && (defined ESP32_SoftwareSerial))
  
      if (oneWire) {
        **frRx = frTx;**     //  Share tx pin. Enable oneWire (half duplex)
        Debug.printf("1-wire half-duplex on pin %d \n", frTx); 
      } else {
      if (set.trmode == ground) {
        Debug.printf("1-wire simplex on tx pin = %d\n", frTx);
      } else { 
        Debug.printf("2-wire on pins rx = %d and tx = %d\n", frRx, frTx);
        if ((set.trmode == air) || (set.trmode == relay)) {
          Debug.println("Use a 2-wire to 1-wire converter for Air and Relay Modes");
        }  
       }  
      }
      frSerial.begin(frBaud, SWSERIAL_8N1, frRx, frTx, frInvert);     // SoftwareSerial
      Debug.println("Using SoftwareSerial for S.Port");
      if (oneWire) {
        **frSerial.enableIntTx(true);**
      }  
  #else  // HardwareSerial
```

Then the switching code looks like this:

```
 void setSPortMode(SPortMode mode) {   
    
  #if (defined TEENSY3X) 
    if(mode == tx && modeNow !=tx) {
      *uartC3 |= 0x20;                 // Switch S.Port into send mode
      modeNow=mode;
    }
    else if(mode == rx && modeNow != rx) {   
      *uartC3 ^= 0x20;                 // Switch S.Port into receive mode
      modeNow=mode;
    }
  #endif

  #if (defined ESP8266) || (defined ESP32) 
      if(mode == tx && modeNow !=tx) { 
        modeNow=mode;
        #if (defined ESP_Onewire) && (defined ESP32_SoftwareSerial)        
          frSerial.enableTx(true);  // Switch S.Port into send mode
        #endif
      }   else 
      if(mode == rx && modeNow != rx) {   
        modeNow=mode; 
        #if (defined ESP_Onewire) && (defined ESP32_SoftwareSerial)                  
          frSerial.enableTx(false);  // disable interrupts on tx pin     
        #endif
      } 
  #endif
  }
```

Sorry if this is a bit long winded. Can you see any obvious misunderstandings on my part? Reading happens successfully, we switch to writing, I can see the pulses on a 'scope, but the FrSky XSR receiver fails to read the 1-wire sent from the ESP32.

Curiously, the base line of the signal on the ESP pin rises about 1.2V during receive, but drops back to zero during transmit. The ESP receives telemetry  just fine though.

[image]

zs6buj, 2020-03-22:
P.S. telemetry speed is 5600 b/s, and the ESP32 is running at 240 MHz

dok-net, 2020-03-22:
57600bps ist asking for trouble on the ESP32. But first and foremost, don't turn on interrupts in onewire mode during rx, what's the point?

dok-net, 2020-03-22:
Please try it on an ESP8266, higher data rates work much better there. I think you've confirmed that onewire works, by using your own test setup, your further troubles with the hardware you really want to connect to are outside my ability to help.

zs6buj, 2020-03-23:
Thanks for the feedback. I can't see where I enabled Rx interrupts? I believe I used the code below, but maybe you noticed something else? 

`   frSerial.enableTx(true);  // Switch S.Port into send mode`

I will persevere and see if I can get it to work :)

dok-net, 2020-03-23:
frSerial.enableIntTx(true);

zs6buj, 2020-03-23:
Is the the RX interrupt?

zs6buj, 2020-03-23:
I saw this in you OneWire example:

```
	swSer1.begin(115200, SWSERIAL_8N1, 12, 12, false, 256);
	swSer1.enableIntTx(true);
	swSer2.begin(115200, SWSERIAL_8N1, 14, 14, false, 256);
	swSer2.enableIntTx(true);
```

dok-net, 2020-03-23:
Thanks, I've fixed it in master.

dok-net, 2020-03-23:
Don't try to loop back swSer1 to swSer2 🙂

zs6buj, 2020-03-23:
ok. Thanks

closed 2020-04-10 by dok-net