plerup/espsoftwareserial #147: Issue with saving received bytes to a String.

kapyaar, 2020-03-28:
I am using the latest release, along with Arduino ESP8266 2.6.3. Trying to talk between two ESP8266 boards, where one is the sender and other is the receiver for the purpose of this test. The receiver sends a string every 5 seconds. Following is for the receiver code.
```

#include <SoftwareSerial.h>
String dataString;
SoftwareSerial swSer;

void setup() {
  Serial.begin(115200);
  swSer.begin(19200, SWSERIAL_8N1, 5, 13, false, 128, 11);
  Serial.println("\nSoftware serial test started");
}

```
I am sending "Hello World!" from the sender and trying to save it as a string or char buffer on the receiver. Tried the following variations of loop() with the results below.

Code 1.

```
void loop() {
  while (swSer.available() > 0) {
    Serial.write(swSer.read());
    yield();
  }
}

```
The receiver can receive and print "Hello World!" onto Serial. This shows that sender is sending data correctly and receiver is getting it correctly.

Code 2.
```
void loop() {
  while (swSer.available() > 0) {
    dataString += (char)swSer.read();
    //Serial.write(swSer.read());
    yield();
  }
  if (dataString.length()) {
    Serial.print(dataString.length());
    Serial.print(">");
    Serial.println(dataString);
    dataString= "";
  }  
}
```

This keeps pringing one character at a time. 
```
1>H
1>e
1>l
1>l
1>o
1> 
1>W
1>o
1>r
1>l
1>d
1>!

```

For some reason, the while (swSer.available()) does not reflect the length correctly? I tried adding different delays in the loop but still the same result. Any thoughts?

devyte, 2020-03-29:
What you observe looks correct to me. The passes through loop are much faster than the serial transmission rate, so there will be several passes over the while() where available() will be false before the next char arrives. Then you get that one char, put it in your dataString, print it out, and clear it immediately. Then you go to doing several passes over while() again waiting for the next char to arrive. 
Your code doesn't match what you want. You need to either define a protocol for yourself, or mark the end of your transmission e. g. send a special byte at the end of your string, and on the reading end detect when it arrives.
Also, you need a timeout in case the end of the string never arrives.

kapyaar, 2020-03-29:
I understand what you are saying, but I am almost certain that swSer.available() never returns anything more than 1 even when there should be more characters in the buffer. 

Also, few things to note.
1. I am using ESP for both the sender and the receiver, same clock speed.
2. I am only checking the receiver every so many seconds to see if any data has been received. Even if there is a difference in clock speed, by this time, all characters should be in the receive buffer of the receiver. right?

devyte, 2020-03-29:
Your problem isn't that while(), it's what you're doing below it. Given how you structured the while, it is very likely that available() will only ever return 0 or 1, which is ok. 
You need to check dataString until the message is complete, and only then process. Example: replace if(dataString.length()) with if(dataString[dataString.length()-1] == '!') or something like that. Plus a timeout. 
Like I said, you should receive and not process until end of msg or timeout.

dok-net, 2020-03-29:
@kapyaar "I am almost certain" in software is almost always part of the problem, not the solution 🙂Please use Stream::readBytes or readString or their companion functions checking for EOL. There's a timeout you can set. #136

kapyaar, 2020-03-29:
@dok-net did nt want to sound obnoxious by saying I can prove it. Will check out readbytes. :)

closed 2020-03-29 by kapyaar