plerup/espsoftwareserial #148: software serial print can hang and cause wdt reset

chaos511, 2020-03-30:
I am using Libary this on a nodemcu and sometimes when the program calls software serial.print it causes the esp to hang for a few seconds then the hardware watchdog resets is there any way to fix this?

code:
  `SoftwareSerial swSer;`
  setup:
    ```swSer.begin(9600, SWSERIAL_8N1, D8, D8, false, 128);
    swSer.enableRx(false);
    swSer.enableTx(true);```
  main:  
  `swSer.print(msg);`

error:
ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1392, room 16
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld

Version:
EspSoftwareSerial@6.8.0

dok-net, 2020-04-10:
On ESP8266, the version of EspSoftwareSerial is inconsequential, please check the README.
Please insert `yield();` into your code immediately before the call to print, I suspect the aggregated time since last yield is too long, therefore the WDT triggers.

dok-net, 2020-05-06:
No new information provided.

closed 2020-05-06 by dok-net