plerup/espsoftwareserial #150: Possible micropython porting?

albertomercurio, 2020-04-08:
Hello,
this library works very well. I wonder if you could port this project in a micropython library too.
Regards,
Alberto

dmascord, 2020-04-08:
Hi Alberto,

I have ported an older version of software serial to Micropython, which has good performance on esp8266, https://github.com/dmascord/micropython as a reference.

Cheers,

Damien

albertomercurio, 2020-04-09:
Wonderful! Is there something like this but for ESP32?

dmascord, 2020-04-09:
The same library should work fine on ESP32, my lab is currently packed in storage so I can't test it sorry.

dmascord, 2020-04-09:
I have merged in the latest master, and attempted to get it to compile for esp32.  Currently it is failing to compile, but it is integrated into the build.  Let me know if you get it working, and I will work on it after Easter break.

dok-net, 2020-04-10:
This discussion has everything to do with Micropython now, nothing with EspSoftwareSerial. I'm closing this. If there are any results of your work that should result in changes to EspSoftwareSerial, please report specifically or provide a PR.

closed 2020-04-10 by dok-net