plerup/espsoftwareserial #152: available() seems don't return value

GianCann, 2020-04-16:
I have used this code:
```
  SoftwareSerial serData;
  serData.begin(115200, SWSERIAL_8N1, RXD2, TXD2, false,64);
  
   while (serData.available() > 0){        
         incomingByte = (byte)serData.read();
         Serial.println("Received:");
         Serial.print(incomingByte);
         if (incomingByte==0x02){
             Serial.println("NACK Received");
            }
       }
[115200nack.zip](https://github.com/plerup/espsoftwareserial/files/4487864/115200nack.zip)

```
But don't read the 0x02 (NACK) byte in incoming buffer.
I've attached a file (rename from .zip to .sr to view with PulveView/Sigork) with the captured data (with hardware logic analyzer) on RX/TX Pin of a ESP32 chip where the 0x02 byte is visibile (in the file, D1 in Rx Line, D0 Tx line).

Where is the problem?

[image]

GianCann, 2020-04-17:
@dok-net what the "wontifx" label mean?

dok-net, 2020-04-17:
@GianCann I'm sorry, my reply wasn't posted for some reasons:
Please show me where in the code or the documentation it states the ESP32 with its fat RTOS supports anywhere near 115200bps in software serial, and I'll change that in an instant.
Cannot work -> won't fix :-)

GianCann, 2020-04-17:
Ok, thank you @dok-net 
So, if I use this code with ESP8266 there are no problem??
(With the ESP32 I can try to slow speed at 57.600)

dok-net, 2020-04-17:
Maximum 57600bps sounds about right for ESP32, IIRC. ESP8266 does fine.
Allow me a question, if you consider using an ESP8266, which has only a single hw uart, could you use one of the three (!) hw uarts on the ESP32 instead of relying on sw serial too quickly there?

GianCann, 2020-04-17:
I tried to use a second serial hardware on ESP32, but I don't find a way to control the TX buffer.

I need to send a various packet of bytes, with a delay of 10ms from packet to packet.

Example:
```
swSer.write(msg2,sizeof(msg2));
    swSer.write(msg3,sizeof(msg3));
    swSer.write(msg4,sizeof(msg4));
    swSer.write(msg5,sizeof(msg5));
    delay(10);
    swSer.write(msg6,sizeof(msg6)); //0x9A,0x20,0x43,0x41,0x4C,0x49,0x42,0x00,0x00,0x00,0x00
    swSer.write(msg7,sizeof(msg7)); //0x9A,0x21,0x00,0x00,0x00,0x00,0x00,0xFF,0x7F,0x47,0x83
    swSer.write(msg8,sizeof(msg8)); //0x9A,0x22,0x00,0x00,0x00,0x00,0x00,0x00,0xC8,0x42,0xCD
    swSer.write(msg9,sizeof(msg9)); //0x9A,0x23,0x00,0x00,0x00,0x00,0x00,0xFF,0x7F,0x47,0x81
    swSer.write(msg10,sizeof(msg10)); //0x92,0x24,0x4E,0x2F,0x41,0x00,0x69
    swSer.write(msg11,sizeof(msg11)); //0x8A,0x25,0x10,0x00,0x40
    swSer.write(msg12,sizeof(msg12)); //0x92,0xA0,0x08,0x01,0x05,0x00,0xC1
    delay(10);
    swSer.write(msg13,sizeof(msg13)); //0x99,0x20,0x44,0x45,0x42,0x55,0x47,0x00,0x00,0x00,0x17

```

If I use the second hardware serial, these bytes are sent without delay and the code finish before the tramission end.

I think that the second serial hardware interface is managed in a separate thread/core.

GianCann, 2020-04-17:
Look the output of logic analyzer:
With Hardware Serial (2nd interface)
[image]

With ESPSoftwareSerial:
[image]

D0: tx output
D1 and D2 are to check the sync/delay(10) in the code.

dok-net, 2020-04-17:
LOL - I love it - ESP32 FreeRTOS........

dok-net, 2020-04-17:
I completely understand your problem - can't help you solve it though.
I am glad if EspSoftwareSerial on ESP8266 can help you achieve your goal, please report in another issue if you find trouble there.

closed 2020-04-17 by dok-net

GianCann, 2020-04-17:
Thank you for the fast support!

GianCann, 2020-04-17:
I can confirm: with the ESP8266 my project works fine at 115200 bps.