plerup/espsoftwareserial #154: MIDI IN trouble

fotosettore, 2020-04-22:
Hi
i'm using your library on ESP32 to connect midi devices
In MIDI OUT all data sent are received and all works fine.
In MIDI IN i cannot receive all data, or better, after some data receiving seems freeze.

My IN configuration is : 
` MIDI_CREATE_INSTANCE(SoftwareSerial, swSerOUT, midiOutput); `
`#define BAUD_RATE_IN 57600`

`swSerIN.begin(BAUD_RATE_IN, SWSERIAL_8N1, 26, 22, false, 256,256); `

In the beginning no data where read and ESP32 seemed deaf.

Then i added this line:
`swSerIN.enableRx(false);`

and the result was this : 

http://www.homoandroidus.com/kemper/kemper_03_on_esp32.PNG

After random calls, all is freeze. sometimes 10, sometime 4 .... random!
I tried to change buffer size, without results.
Please note that the same call software (that is a little library to dialogue with kemper midi device) works fine (i tried continuosly 24 hours) using hardware serial in MEGA2560. 
However for development needs i must use ESP32, so i need to solve this issue using that device.
Last : I tried to use HardwareSerial on ESP32, but it is deaf too.

Any help is appreciated.

dok-net, 2020-05-06:
If hardware serial on ESP32 shows the same behavior, I'm good with it. Please try lower bps if your device allows it, even just for testing. Thanks.

dok-net, 2020-05-16:
No feedback received.

closed 2020-05-16 by dok-net