plerup/espsoftwareserial #155: Line break detection

Chreetz, 2020-04-28:
Is there any way to add line break detection? 
I have an application that is sending line breaks to indicate the start of a new datagram. The line is held low (ttl-level) for longer period of time than it takes to send a regular character (4800 8N1). Sometimes the line is held low for 9 bits, sometimes even longer. It should result in an illegal stop-bit, but I cannot figure out where to add this test.
Any help would be appriciated,
Chris

dok-net, 2020-05-06:
Break detection was not a design objective. YGIAGAM.
There's no such thing as illegal-stop bit reporting in the implementation, you could tinker with parity, though.

dok-net, 2020-05-16:
You're invited to prepare an outline or a PR for such an enhancment, otherwise, there's no known interest in break detection, also, please compare to HW serial, you could exchange both for each other in many use cases.

closed 2020-05-16 by dok-net