plerup/espsoftwareserial #156: [ESP32] SoftwareSerial in combination with SPIFFS crashes

bart, 2020-05-01:
Hey everyone,

I want to use the SoftwareSerial package to log some GPS data using the TinyGPS library. For fast data writing I like to use SPIFFS but it looks like that it doesn't work in combination with SoftwareSerial. The ESP32 just crashes and reboots over and over again.

Here is a sample code that fails:

```
#include "FS.h"
#include "SPIFFS.h"
#include <SoftwareSerial.h>

#define RX_PIN 18
#define TX_PIN 19
#define BAUD_RATE 9600

SoftwareSerial ss(RX_PIN, TX_PIN);
String filepath = "/log.txt";
File file;

void setup() {
  ss.begin(BAUD_RATE);

  SPIFFS.begin();
  file = SPIFFS.open(filepath, FILE_WRITE);
}

void loop() {
  file.println(String(millis()) + ";" + String(esp_random()) + ";" + String(esp_random()) + ";" + String(esp_random()));

  delay(20);
}
```
When I remove line `ss.begin(BAUD_RATE);` it works fine.
Any ideas or suggestions? Thanks a lot!

bart, 2020-05-01:
Here is the decoded exception:

```
PC: 0x400f14bc: circular_queue ::push(unsigned int&&) at /Users/bart/Documents/Arduino/libraries/EspSoftwareSerial/src/circular_queue/circular_queue.h line 269
EXCVADDR: 0x00000000

Decoding stack results
0x400f14bc: circular_queue ::push(unsigned int&&) at /Users/bart/Documents/Arduino/libraries/EspSoftwareSerial/src/circular_queue/circular_queue.h line 269
0x4008114d: __onPinInterrupt at /Users/bart/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/esp32-hal-gpio.c line 218
0x4008d62b: esp_rom_spiflash_read_status at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c line 261
0x4008d662: esp_rom_spiflash_wait_idle at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c line 35
0x4008d701: esp_rom_spiflash_erase_sector_internal at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c line 121
0x4008db96: esp_rom_spiflash_erase_sector at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c line 451
0x40086b41: spi_flash_erase_range at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c line 237
0x400efd4d: esp_partition_erase_range at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/partition.c line 294
0x400d8955: spiffs_api_erase at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs_api.c line 58
0x400da202: spiffs_erase_block at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c line 236
0x400d8d5d: spiffs_gc_erase_block at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_gc.c line 15
0x400d989e: spiffs_gc_check at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_gc.c line 168
0x400dafbd: spiffs_object_append at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c line 1219
0x400d77ed: spiffs_hydro_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_hydrogen.c line 450
0x400d80f5: SPIFFS_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_hydrogen.c line 519
0x400d7486: vfs_spiffs_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spiffs/esp_spiffs.c line 466
0x400df742: esp_vfs_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/vfs/vfs.c line 420
0x400e0ad4: _fwrite_r at ../../../.././newlib/libc/stdio/fwrite.c line 170
0x400e0b1d: fwrite at ../../../.././newlib/libc/stdio/fwrite.c line 211
0x400d145e: VFSFileImpl::write(unsigned char const*, unsigned int) at /Users/bart/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/FS/src/vfs_api.cpp line 325
0x400f128d: fs::File::write(unsigned char const*, unsigned int) at /Users/bart/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/FS/src/FS.cpp line 50
0x400d3079: Print::print(String const&) at /Users/bart/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/Print.h line 71
0x400d309d: Print::println(String const&) at /Users/bart/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/Print.cpp line 175
0x400d1019: loop() at /var/folders/2j/dsw6y2kx2ks2x4_7qwxqyq080000gn/T/arduino_modified_sketch_88462/sketch_may01a.ino line 21
0x400d3749: loopTask(void*) at /Users/bart/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/main.cpp line 19
0x400882d5: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

bart, 2020-05-04:
I solved the problem by using hardware serial instead which is more a workaround but closes this issue for me.

closed 2020-05-04 by bart

CurtRod, 2020-06-06:
I have the same issue using SoftwareSerial in combination with SPIFFS on ESP32. The Stack Trace is nearly the same and ends with:
0x400f14bc: circular_queue ::push(unsigned int&&) at /Users/bart/Documents/Arduino/libraries/EspSoftwareSerial/src/circular_queue/circular_queue.h line 269

Any suggestions from the developers here?