plerup/espsoftwareserial #159: Q: required amount of heap?

selangh, 2020-05-09:
Hello,

in the past I used this [version](https://github.com/Trialcommand/ESP8266-Modbus-RTU-Master/tree/master/SoftwareSerial) of SoftwareSerial + ModbusMaster in my project (ESP8266 D1mini) for reading values from a modbus device. Recently I switched to the latest version of SoftwareSerial (6.8.1 installed from the libraries in vscode and platformio). Now I have 6kb less heap than before. I can reproduce my old situation of heap usage just by replacing the 6.8.1 version with the old one from the link above.

I´d like to maintain only my own code in my repo without third party stuff. So I would be very happy to reduce the heap usage of the recent SoftwareSerial library about 6kb? Will this be possible somehow? My implementation uses two instances of SoftwareSerial. In the old version of the library I specified 512 Byte buffer size for each. In the most recent version I didn´t changed anything because it seems to me that it already uses a much smaller buffer size.

The code looks something like this:
```c
#include "SoftwareSerial.h"
#include "ModbusMaster.h"

SoftwareSerial ModbusSerial(12, 14, false);
ModbusSerial.begin(19200);

ModbusMaster modbus;
uint8_t ModBusSlaveID = 10;
modbus.begin(ModBusSlaveID, ModbusSerial);
...
```

Thanks in advance
Sebastian

dok-net, 2020-05-09:
You should really take you own time first to read the docs, learn about your buffer size needs, mingle with other users, before wasting developer time asking for help in this development issue tracking system.It's less than 600 bytes difference in heap usage.

selangh, 2020-05-10:
Read the docs - indeed... that helped me out. In comparision to the old version that simply worked out of the box this one needed a little adjustment in buffersizes. Your

> "It's less than 600 bytes difference in heap usage." 

was a good hint. I almost reached this value. So I am happy for now.

closed 2020-05-10 by selangh