plerup/espsoftwareserial #160: Can SoftwareSerial run 2 ports and tx/rx simultaneously

andyb2000, 2020-05-14:
Sorry, not so much a bug/report but I wasn't aware of anywhere else to ask this.
Does this have the same limitation as the 'core' softserial library that cannot tx+rx at the same time?

I'm trying to write an 'main-in-the-middle' serial system on ESP8266 to read binary data at 9600-b-n-1 from one softserial and write it back to the other, so code like this:

`SoftwareSerial swSer1(12, 0, false); // RX port then TX port
SoftwareSerial swSer2(13, 4, false); // RX port then TX port`

`void loop() {
      uint8_t rc;
unsigned long starttime = millis();
    while (swSer2.available()) {      
      runtime = millis();
      rc = swSer2.read();
      swSer1.enableTx(true);
      swSer1.print(rc,BIN);
      swSer1.enableTx(false);
   };
    };`

Is this possible in espsoftwareserial?

dok-net, 2020-05-14:
At 9600bps, I think it's safe to invite you to turn your code into something that runs, test and report here. Have fun and good luck!

dok-net, 2020-05-14:
BTW, wherever that came from, all that enableTx I'd really redundant in non-singlewire ops, please drop that.

andyb2000, 2020-05-15:
ok, so drop enableTX, I didn't see the notes on that being for singlewire.
I'm trying this repeatedly using different code and seems to just not work right but I cannot determine why, so ideally I'm looking for a confirmation this should work.

So my code I'm trying:

`#include <SoftwareSerial.h> // This is the one for ESP8266 specifically
SoftwareSerial swSer1(11, 12, false); // RX port then TX port
SoftwareSerial swSer2(9, 10, false); // RX port then TX port
#define MAX_MILLIS_TO_WAIT 8000  //How long to wait for serial data to avoid hanging code
byte startByte = 188;                     // <- start byte of serial 10111100
byte endByte = 253;                       // <- stop byte of serial 11111101
boolean newData = false;
byte ch;
byte ser_indata[10]; // byte array for serial binary data

void setup() {
  Serial.begin(115200);
  delay(3000);
  Serial.println(F(" Debug serial binary output"));
  swSer1.begin(9600,SWSERIAL_8N1);
  swSer2.begin(9600,SWSERIAL_8N1);
}

void loop() {
  static boolean recvInProgress = false;
      static byte ndx = 0;
      uint8_t rc;
    unsigned long starttime = millis();
    while (swSer1.available() > 0 && newData == false && ((millis() - starttime) < MAX_MILLIS_TO_WAIT)) {      
      rc = swSer1.read();
      Serial.println(rc,BIN);
      swSer2.write(rc,BIN);
      if (recvInProgress == true) {
        if (rc != endByte) {
          ser_indata[ndx] = rc;
          ndx++;
        } else {
          end_byte=runtime;
          recvInProgress = false;
          ndx = 0;
          newData = true;
          swSer1.flush(); // flush now we have a valid pack of data?
        }
      } else if (rc == startByte) {
        start_byte=runtime;
        recvInProgress = true;
      };
    };
    if (newData == true) {
      Serial.println("Data found:");
      for(int i = 0; i < sizeof(ser_indata); i++)
      {
          Serial.print(ser_indata[i]);
          Serial.print("  =  ");
         Serial.println(ser_indata[i],BIN);
      };
      newData = false;
    };
}`

dok-net, 2020-05-16:
"Builds" is the step before runs, I highly doubt your sketch builds in your place. Now seriously, this is not a MCVE. Loopback works, duplex in software on a single core obviously has it's limits, but your sketch appears to be all about half-duplex, so no known issues with that.
To be sure, this is the development repository, not the place for user support IMHO - I honestly feels spammed by questions at this level. No offense, but this clutters my mailbox, and I can't answer questions at this level. Please head on over to stackexchange or whatever, a place where user to user support is happening.

closed 2020-05-16 by dok-net