plerup/espsoftwareserial #162: availableForWrite() should return number of bytes

jandrassy, 2020-05-24:
https://www.arduino.cc/reference/en/language/functions/communication/serial/availableforwrite

> availableForWrite()
> Description
> 
> Get the number of bytes (characters) available for writing in the serial buffer without blocking the write operation.
>

dok-net, 2020-05-24:
What is the exact nature of the issue you are reporting? That function exists in the current release. The interpretation of "without blocking" is debatable.

jandrassy, 2020-05-25:
current implementation only returns m_txValid (in an obscure way)

dok-net, 2020-05-25:
Please detail what exactly is the deviation from documented or expected behavior. If you have better code, provide a PR for public review. Otherwise, just close this item.

jandrassy, 2020-05-25:
I see there is no internal buffer so sorry the implementation is right. It can't do anything better.

closed 2020-05-25 by jandrassy