plerup/espsoftwareserial #164: ESP8266 continually resets receiving serial input with a 2KB soft serial buffer

Jasoroony, 2020-05-30:
I have an ESP8266-01 project that uses the main Serial port for adding WiFi features to a host device, so I'm using ESP software serial to connect a GPS device as a 2nd function (the host device only has 1 available serial port, so I'm attempting to multiplex the WiFi & GPS data together within the ESP8266).

This is working when I set the software serial receive buffer to 512 bytes.  It sometimes works when I set the buffer size to 1024 bytes depending on how much memory I'm using.  It never works when I attempt to use 2048 bytes for the software serial buffer size.  I want at least 2KB because the WiFi routines are sometimes slow, and I don't want to drop the GPS data while it's waiting on WiFi connects.

The GPS device will be configured for 115200 (set by a command on the Serial1 / IO02 hardware transmit only serial port of the ESP8266 - which is also working Ok).  Initially though, the GPS boots @ 9600 BAUD and transmits data (~150 odd bytes) once a second.  I suspect this data is causing the ESP to reboot with a software serial 2KB receive buffer, as it otherwise seems stable when nothing is being transmitted to the ESP IO00 pin.  I'm not sure if the differing BAUD rates is the problem?  I wouldn't have thought so though.

Here is some basic code that constantly runs the startup LED flashing routine (so I know it's rebooting at about the 1 second intervals the GPS device outputs its location data):

```
#include <SoftwareSerial.h>

#define BAUD_COM 1000000
#define BAUD_GPS 115200
#define SSSIZE (2*1024)

static SoftwareSerial swSer1;

void setup() {
    pinMode (2, OUTPUT);
    digitalWrite (2, 0);
    delay (500);
    digitalWrite (2, 1);
    delay (500);
    digitalWrite (2, 0);
    delay (500);
    pinMode (2, INPUT);
    
    Serial.begin (BAUD_COM);

    swSer1.begin (BAUD_GPS, SWSERIAL_8N1, 0, 5, false, SSSIZE);
    swSer1.enableIntTx (false);
}

void loop() {
    if (Serial.available ())
        Serial.read ();
    if (swSer1.available ())
        swSer1.read ();
}
```

Jasoroony, 2020-06-01:
Being a public holiday today where I am, I had time to look into this.  It turns out, the library's begin method allocates 1 * requested buffer size as expected, plus 1/8th of that if parityMode is true, and 10 * requested buffer size * sizeof (uint32) for the real ISR working buffer.

So this implementation of SoftwareSerial uses ~45 times the amount of memory needed for the desired end result.

@1024 it uses ~45KB (which sometimes works)
@2048 is uses ~95KB (which will never work)

Mystery solved.  So I can't see any way for my project to not drop GPS data.  When the GPS chip has a lock on a position, it can output >512 bytes a second.  The WiFi connect call can take multiple seconds to complete which will almost always overflow the buffers I can allocate.

I guess this is not a valid issue for this particular library (I'm not inclined to click close though).

Incidentally, in the begin method, I don't understand why allocating the buffer and parityBuffer vars checks if bufCapacity is 0 (which then uses defaults) but isrBuffer doesn't do the same check and use 64 for its default.  Instead, it will multiply the request by 0 and create a zero sized buffer.

Jasoroony, 2020-06-04:
A few days have passed.  Over this time, I've been thinking about the direction that this library has taken, also because I've found one that works for me quite a bit better from issue #159 ([https://github.com/Trialcommand/ESP8266-Modbus-RTU-Master/tree/master/SoftwareSerial](url)) I'm seeing the real issue being solved here.  That's good, WiFi is a bit flakey with the old lib, and is better with this one especially as the requested BAUD rate is lowered.

My idea is, set the character buffer I need (that can't change, it needs to be 2-8KB) but reduce the uint32 interrupt buffer down to something more acceptable, which is already configurable.  Somehow (gee, this C++ code has become complex vs that older lib to the point it hurts my head) make a settable "reliable" flag that causes the interrupt routine to also eat enough uint32 bit-times to put one byte into the character buffer when the uint32 buffer is close to full.  A fallback, if you like.  I wonder at what (low) BAUD rate you could do that, process the 10 required bits when needed, or process 1 if the routine(s) allows it??, within the timing needed before the next start bit arrives?

Then I noticed there are two interrupt routines, and above 74880 BAUD it uses an alternate function, which falls back to the old library timing algorithm and no longer tries to improve WiFi.  Well, since I'm already thinking about dropping the BAUD rate, that's maybe not so bad, but all of this is too much for me to do with how the C++ code looks, it's so much more complex than the old code now.

dok-net, 2020-06-07:
> Incidentally, in the begin method, I don't understand why allocating the buffer and parityBuffer vars checks if bufCapacity is 0 (which then uses defaults) but isrBuffer doesn't do the same check and use 64 for its default. Instead, it will multiply the request by 0 and create a zero sized buffer.

Thanks for pointing out this one, I am looking into it. The isrBuffer size calculation should be able to take SoftwareSerialConfig into consideration, too.

dok-net, 2020-06-07:
Otherwise, doing anything in the ISR of what you are describing is not going to happen. Which WiFi connect is synchronously taking so much time, are you certain, that there is no async alternative available? You can run either `available()` from your `setup()` or `loop()` code, or `perform_work()`.

dok-net, 2020-06-07:
I think the question is answered now.

closed 2020-06-07 by dok-net