plerup/espsoftwareserial #165: SoftwareSerial crashes in combination with SPIFFS (ESP32)

CurtRod, 2020-06-06:
I use SoftwareSerial with an ESP32 in combination with SPIFFS to write some log files. In some cases the ESP crashes with the following stack trace. It happens while writing SPIFFS a short time after SoftSerial communication. I use The Pins 22 (RX) and 21 (TX):

PC: 0x4019334c: circular_queue ::push(unsigned int&&) at C:\Users\curt\.platformio\lib\EspSoftwareSerial_ID168\src\circular_queue/circular_queue.h line 269
EXCVADDR: 0x00000000

Decoding stack results
0x4019334c: circular_queue ::push(unsigned int&&) at C:\Users\curt\.platformio\lib\EspSoftwareSerial_ID168\src\circular_queue/circular_queue.h line 269
0x40081a29: __onPinInterrupt at C:\Users\curt\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-gpio.c line 218
0x40092e2f: esp_rom_spiflash_read_status at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c line 261
0x40092e66: esp_rom_spiflash_wait_idle at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c line 35
0x40092f05: esp_rom_spiflash_erase_sector_internal at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c line 121
0x4009339a: esp_rom_spiflash_erase_sector at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c line 451

There is already an issue (closed) detailing the problem.: #156 

Any suggestions or ideas to solve this?

Thanks in advance.

dok-net, 2020-06-07:
@CurtRod If you want me to consider looking into this any further, please provide an MCVE. Otherwise, some quick research indicates that SPIFFS may be having some kind of issues with concurrent interrupts. If your use cases allows for dropping received data during SPIFFS access, could you disable interrupts around those calls and see what happens?

CurtRod, 2020-06-13:
Thank you for the quick reaction and the hint. The crash always happened shortly after the log file was written when an RX interrupt was triggered. I have now deactivated the interrupts around the SPIFFS writes and added a short delay before and after. So far it works for me!

closed 2020-06-13 by CurtRod