plerup/espsoftwareserial #167: Speed limited to 14400baud with ESP32 DEVKIT

petermill, 2020-06-08:
First some background.
I am a French electronic engineer, now retired and "playing" with Model railways. 
I use the OpenDCC and particularly the BiDiB. [http://www.bidib.org/index_e.html](url).

I am working on an android terminal to drive the locos on rail. The aim is to interface the BiDiBus via Wlan. so the interest in IOT and specially ESP32. I bought myself a Dev Kit ESP32

As BiDiBus uses serial transmission with 9 bits 500 000bauds (ATmel), I try to use the espsoftware serial. I have added the ESP-IDF onto Arduino IDE.

First, the example given sewsertest, don't compile with ESP32 
error: "D5 is not defined".

Second: In order to understand and test I have created the attached SW.
[UART_IT.ino.txt](https://github.com/plerup/espsoftwareserial/files/4745968/UART_IT.ino.txt) Rename in .ino (the site don't accept file ending in .ino).

I test the result with a logic analyser (Salae).
Until 14 400 bauds SW works fine; Above 14400 bauds, starting at 28800bauds all the " swSer.write(ch);" are wrong.
Is it normal ?

Pierre

dok-net, 2020-06-09:
You will never get 500kbps out of softwareserial on the ESP32.
Insofar as 14400bps works in your tests, that points in the direction that you are not hitting any defects in EspSoftwareSerial or your other code. The examples included with EspSoftwareSerial all run in my configuration, from them, you can see that the the trivial swsertest is set to 57600bps, but the more demanding loopback and repeater sketches are set to 19200bps. You can take that as a hint that I wasn't able to get stable results above that bitrate.

> First, the example given sewsertest, don't compile with ESP32 error: "D5 is not defined".

There are ESP32 boards that have identical pin layout to the Wemos D1s minis, I'm using these in my tests. I keep thinking that it's one of the more obvious things in the sketches to define the rx and tx according to the actual setup. I'll try to keep that in the back of my mind, until the next time I have to touch that code.

Having said all that, I don't know if there is anything that can be done for you. If you need the high bitrate, you'll have to research if and how the HW UARTs' parity modes allow for setting mark and space, plus how to read parity on receive, instead of words with bad parity getting discarded silently.

TL;DR:
> Is it normal ?

Yes.

petermill, 2020-06-09:
The API in ESP32 esp-idf are real convenient but they hide too many possibilities.
You are right, since the HW works at 500 000 bauds, I need to go in detail to see how to add mark and space and stop "discarded silently bad parity"

closed 2020-06-09 by petermill