plerup/espsoftwareserial #169: Not Working

notsolowki, 2020-06-11:
This library is not working with my DFROBOT CO2 Sensor. the sensor uses UART and on the arduino it works fine. the uart is 3.3v. Why does this not work on esp? i uses the examples that come with the library as reference but its still a no go.

dok-net, 2020-06-11:
Very bad library.
MCVE?
I'm sorry but my crystal ball is out for maintenance.

closed 2020-06-11 by dok-net

notsolowki, 2020-06-11:
What does MCVE mean? this is the code i use for the arduino UNO. Very bad library? u mean this library right? because this is the one im using. 

`#include <SoftwareSerial.h>
SoftwareSerial mySerial(10, 11); // RX, TX
unsigned char hexdata[9] = {0xFF,0x01,0x86,0x00,0x00,0x00,0x00,0x00,0x79}; //Read the gas density command /Don't change the order
void setup() {

  Serial.begin(9600);
  while (!Serial) {

  }
  mySerial.begin(9600);

}

void loop() {
   mySerial.write(hexdata,9);
   delay(500);

 for(int i=0,j=0;i<9;i++)
 {
  if (mySerial.available()>0)
  {
     long hi,lo,CO2;
     int ch=mySerial.read();

    if(i==2){     hi=ch;   }   //High concentration
    if(i==3){     lo=ch;   }   //Low concentration
    if(i==8) {
               CO2=hi*256+lo;  //CO2 concentration
      Serial.print("CO2 concentration: ");
      Serial.print(CO2);
      Serial.println("ppm");
               }

  }

 }

 }`

notsolowki, 2020-06-11:
I am using D5,D6 on the nodemcu. here is the code i use on the nodemcu,

``
#include <SoftwareSerial.h>

SoftwareSerial mySerial(14, 12); // RX, TX
unsigned char hexdata[9] = {0xFF,0x01,0x86,0x00,0x00,0x00,0x00,0x00,0x79}; //Read the gas density command /Don't change the order
void setup() {

  Serial.begin(9600);
  while (!Serial) {

  }
  
  mySerial.begin(9600);
Serial.println("hi");
}

void loop() {
   mySerial.write(hexdata,9);
   delay(500);

 for(int i=0,j=0;i<9;i++)
 {
  if (mySerial.available()>0)
  {
     long hi,lo,CO2;
     int ch=mySerial.read();

    if(i==2){     hi=ch;   }   //High concentration
    if(i==3){     lo=ch;   }   //Low concentration
    if(i==8) {
               CO2=hi*256+lo;  //CO2 concentration
      Serial.print("CO2 concentration: ");
      Serial.print(CO2);
      Serial.println("ppm");
               }

  }

 }

 }
``

i dont know how to be any more clear

notsolowki, 2020-06-11:
From [https://www.dfrobot.com/product-1565.html](url) i am using sen0220.

I never have used software serial on ESP before. but i must be missing something because it does not work at all

notsolowki, 2020-06-11:
okay @dok-net first, i did not change whatever EspSoftwareSerial.begin() your talking about i have no clue where you see this at all in the first place. 

second my only change is the pin assignment did you even look at the code i posted?

dok-net, 2020-06-11:
> work at all

Perhaps you are trying to convey the information that it's not even compiling, or does it miraculously? Which version of Arduino core for ESP8266 are you using, then?

notsolowki, 2020-06-11:
> > work at all
> 
> Perhaps you are trying to convey the information that it's not even compiling, or does it miraculously? Which version of Arduino core for ESP8266 are you using, then?

no it compiles and uploads fine, i am using the latest version

dok-net, 2020-06-11:
"okay" ;-) I am sure you are in the wrong place for asking this type of elementary question, this is the developer support issue tracking, questions at this level clutter my inbox, please forgive me, that I am not so amused. Still, I'm trying to help.

notsolowki, 2020-06-11:
the only thing i changed in the code is, the pin assignment as follows, 

i can confirm in using the correct library i downloaded from here by checking the compiler output. 

```
/***************************************************
* Infrared CO2 Sensor 0-50000ppm(Wide Range)
* ****************************************************
* The follow example is used to detect CO2 concentration.

* @author lg.gang(lg.gang@qq.com)
* @version  V1.0
* @date  2016-6-6

* GNU Lesser General Public License.
* See <http://www.gnu.org/licenses/> for details.
* All above must be included in any redistribution
* ****************************************************/
#include <SoftwareSerial.h>
SoftwareSerial mySerial(14, 12); // RX, TX
unsigned char hexdata[9] = {0xFF,0x01,0x86,0x00,0x00,0x00,0x00,0x00,0x79}; //Read the gas density command /Don't change the order
void setup() {

  Serial.begin(9600);
  while (!Serial) {

  }
  mySerial.begin(9600);

}

void loop() {
   mySerial.write(hexdata,9);
   delay(500);

 for(int i=0,j=0;i<9;i++)
 {
  if (mySerial.available()>0)
  {
     long hi,lo,CO2;
     int ch=mySerial.read();

    if(i==2){     hi=ch;   }   //High concentration
    if(i==3){     lo=ch;   }   //Low concentration
    if(i==8) {
               CO2=hi*256+lo;  //CO2 concentration
      Serial.print("CO2 concentration: ");
      Serial.print(CO2);
      Serial.println("ppm");
               }

  }

 }

 }
```

notsolowki, 2020-06-11:
for simplicity here is a pinout relating to Arduino IDE for the nodemcu

[image]

Also thank you again for helping

dok-net, 2020-06-11:
I've double checked the code, this is just the most basic initialization and write-delay-read there is.
Could it possibly be that you are not supplying the sensor the voltage and current it needs to operate properly?

notsolowki, 2020-06-11:
Please do you have any idea what is going wrong to cause this not to work? according to dfrobot the UART is 0-3.3v so this should be fine for esp?  This is not the first place i requested help with this issue. I have tried the obvious, multiple esp modules, checking the wiring etc. I am supplying the sensor with power. i have a $1,300 programmable DC power supply as my power supply. Yes i have connected the ground between the two.

dok-net, 2020-06-11:
Alright, a smart thing to try whenever there is
- only diagnostic output on the USB UART
- only a single serial device needs to be attached
is to swap HW and SW serials.

notsolowki, 2020-06-11:
Please forgive me for asking a silly question like this but, would this code revision get it operating on the HW serial? 

```

/***************************************************
* Infrared CO2 Sensor 0-50000ppm(Wide Range)
* ****************************************************
* The follow example is used to detect CO2 concentration.

* @author lg.gang(lg.gang@qq.com)
* @version  V1.0
* @date  2016-6-6

* GNU Lesser General Public License.
* See <http://www.gnu.org/licenses/> for details.
* All above must be included in any redistribution
* ****************************************************/
#include <SoftwareSerial.h>

//SoftwareSerial mySerial(14, 12); // RX, TX
unsigned char hexdata[9] = {0xFF,0x01,0x86,0x00,0x00,0x00,0x00,0x00,0x79}; //Read the gas density command /Don't change the order
void setup() {

  Serial.begin(9600);
  while (!Serial) {

  }
  
  Serial.begin(9600);
Serial.println("hi");
}

void loop() {
   Serial.write(hexdata,9);
   delay(500);

 for(int i=0,j=0;i<9;i++)
 {
  if (Serial.available()>0)
  {
     long hi,lo,CO2;
     int ch=Serial.read();

    if(i==2){     hi=ch;   }   //High concentration
    if(i==3){     lo=ch;   }   //Low concentration
    if(i==8) {
               CO2=hi*256+lo;  //CO2 concentration
      Serial.print("CO2 concentration: ");
      Serial.print(CO2);
      Serial.println("ppm");
               }

  }

 }

 }
```

dok-net, 2020-06-11:
No, it's not THAT simple :-)
Use `Serial.swap();` to redirect the HW UART to D7 and D8 - and connect your sensor there.
But you also want the SW serial to speak to TX, so set up the pins such that it matches up to that.
Then, of course, in the sketch exchange Serial and mySerial.

notsolowki, 2020-06-11:
Okay this is what i have come up with but im not very confident about its validity. I never really worked with UART on the esp and until today i have never heard of serial.swap. 

I think they way i have set this up i have setup the SoftwareSerial on pin 13 which is the TX pin for HW serial after Serial.swap()? So in this case im writing hexdata out of pin 13. please forgive me i know you are probably very busy.

```
/***************************************************
  Infrared CO2 Sensor 0-50000ppm(Wide Range)
* ****************************************************
  The follow example is used to detect CO2 concentration.

  @author lg.gang(lg.gang@qq.com)
  @version  V1.0
  @date  2016-6-6

  GNU Lesser General Public License.
  See <http://www.gnu.org/licenses/> for details.
  All above must be included in any redistribution
* ****************************************************/
#include <SoftwareSerial.h>
SoftwareSerial mySerial(10, 11); // RX, TX
unsigned char hexdata[9] = {0xFF, 0x01, 0x86, 0x00, 0x00, 0x00, 0x00, 0x00, 0x79}; //Read the gas density command /Don't change the order
void setup() {

  Serial.begin(9600);
  Serial.swap();
  while (!Serial) {

  }
  mySerial.begin(9600, SWSERIAL_8N1, -1, 13);

}

void loop() {
  mySerial.write(hexdata, 9);
  delay(500);

  for (int i = 0, j = 0; i < 9; i++)
  {
    if (Serial.available() > 0)
    {
      long hi, lo, CO2;
      int ch = Serial.read();

      if (i == 2) {
        hi = ch;    //High concentration
      }
      if (i == 3) {
        lo = ch;    //Low concentration
      }
      if (i == 8) {
        CO2 = hi * 256 + lo; //CO2 concentration
        Serial.print("CO2 concentration: ");
        Serial.print(CO2);
        Serial.println("ppm");
      }

    }

  }

}
```

notsolowki, 2020-06-11:
I think i did this part right,
```
Use Serial.swap(); to redirect the HW UART to D7 (rx) and D8 (tx) - and connect your sensor there.
```

Here im not sure about,

```
But you also want the SW serial to speak to TX, ultimately the USB UART, so set up the pins such that it matches up to that, like

mySerial.begin(9600, SWSERIAL_8N1, -1, TX);
Then, of course, in the sketch exchange Serial and mySerial. And, as you did, print some diagnostic boot line from setup() to prove that SW serial is correctly sending to your PC via USB.
```

The USB serial will still work with Serial.swap? I tried mySerial.println("hello"); but nothing prints from the console.

dok-net, 2020-06-11:
Just try the sketch in your question, I've edited that.

notsolowki, 2020-06-11:
Okay thank you soo much, i do indeed see "Detecting CO2 concentration" print in my serial console ONLY if my sensor is not plugged in to the tx pin on D8.

dok-net, 2020-06-11:
What do you mean? If you connect the sensor to D7 and D8, not TX and RX, which are for the USB UART which now gets handled by EspSoftwareSerial, the sketch logs nothing else but the "welcome" message? So the sensor itself still does not get read?

notsolowki, 2020-06-11:
I have sensor TX connected to D7 and sensor RX connected to D8. in this configuration nothing at all is printed from the console. if i unplug the D7 and D8 then i see "Detecting CO2 concentration"

dok-net, 2020-06-11:
That doesn't sound like a software problem...

notsolowki, 2020-06-11:
I cant even upload a sketch with anything connected to d7/d8

dok-net, 2020-06-11:
AFAIK D7/D8 are not involved in reset, boot, flash, anything elementary. This is not the first attempt at `swap()` for me, never had any problem. Whatever is happening to you, it's not related to EspSoftwareSerial at this stage.

notsolowki, 2020-06-11:
Interesting enough, on my oscilloscope i can see data. in the past 5 minutes 2 readings have came through from the serial console. Do i need maybe some diode or pulldown resistor ?

notsolowki, 2020-06-11:
With a 10k pulldown i get all kinds of crazy readings,

```
10:49:28.104 -> CO2 concentration: 14344ppm
10:49:28.624 -> CO2 concentration: 14336ppm
10:49:29.142 -> CO2 concentration: 0ppm
10:49:29.693 -> CO2 concentration: 4ppm
10:49:30.214 -> CO2 concentration: 0ppm
10:49:30.735 -> CO2 concentration: 65280ppm
10:49:31.249 -> CO2 concentration: 30ppm
10:49:31.804 -> CO2 concentration: 0ppm
10:49:32.322 -> CO2 concentration: 65024ppm
10:49:32.838 -> CO2 concentration: 64ppm
10:49:33.393 -> CO2 concentration: 0ppm
10:49:33.886 -> CO2 concentration: 63488ppm
10:49:34.446 -> CO2 concentration: 0ppm
10:49:34.964 -> CO2 concentration: 0ppm
10:49:35.487 -> CO2 concentration: 0ppm
10:49:36.006 -> CO2 concentration: 0ppm
10:49:36.530 -> CO2 concentration: 8ppm
10:49:37.050 -> CO2 concentration: 0ppm
10:49:37.611 -> CO2 concentration: 0ppm
10:49:38.130 -> CO2 concentration: 0ppm
10:49:38.650 -> CO2 concentration: 34304ppm
```

dok-net, 2020-06-11:
I think we have established that the original cause of failure, behind this issue, is something entirely different than EspSoftwareSerial. I'm glad I could help you learn about Serial::swap(), that's always handy to check if some failure actually depends on EspSoftwareSerial and can be fixed by using HW serial. Otherwise, though, I'll have to leave this to fix on your own now. Your are welcome to report you final solution here, for anyone else looking into how to get that particular sensor working with an ESP.

notsolowki, 2020-06-11:
Okay i understand thank you

notsolowki, 2020-06-11:
While fooling around i decided to put a diode on D8. Well now it works.

dok-net, 2020-06-11:
Would you like to try going back to HW handling the USB, and SW serial handling the CO2 sensor, and try your diode hack? You wouldn't own any other serial device you could attach for testing?

notsolowki, 2020-06-11:
I have some cp2102 adapters. I could run the SoftwareSerial. Really all id have to do is change the pin numbers right? in other words, this should work right?

```
/***************************************************
* Infrared CO2 Sensor 0-50000ppm(Wide Range)
* ****************************************************
* The follow example is used to detect CO2 concentration.

* @author lg.gang(lg.gang@qq.com)
* @version  V1.0
* @date  2016-6-6

* GNU Lesser General Public License.
* See <http://www.gnu.org/licenses/> for details.
* All above must be included in any redistribution
* ****************************************************/
#include <SoftwareSerial.h>
SoftwareSerial mySerial(14, 12); // RX, TX
unsigned char hexdata[9] = {0xFF,0x01,0x86,0x00,0x00,0x00,0x00,0x00,0x79}; //Read the gas density command /Don't change the order
void setup() {

  Serial.begin(9600);
  while (!Serial) {

  }
  mySerial.begin(9600);

}

void loop() {
   mySerial.write(hexdata,9);
   delay(500);

 for(int i=0,j=0;i<9;i++)
 {
  if (mySerial.available()>0)
  {
     long hi,lo,CO2;
     int ch=mySerial.read();

    if(i==2){     hi=ch;   }   //High concentration
    if(i==3){     lo=ch;   }   //Low concentration
    if(i==8) {
               CO2=hi*256+lo;  //CO2 concentration
      Serial.print("CO2 concentration: ");
      Serial.print(CO2);
      Serial.println("ppm");
               }

  }

 }

 }
```

dok-net, 2020-06-11:
Looks good to me.

dok-net, 2020-06-11:
You've two diodes now, one each on D7 and D8?

dok-net, 2020-06-11:
Without a circuit diagram I am not sure what you mean.

notsolowki, 2020-06-11:
I hope this picture can better help its the best i can do right now.

[image]

dok-net, 2020-06-11:
The diode on D7 seems superfluous, don't you agree?

notsolowki, 2020-06-11:
I do now. thank you. it seems to work fine without the diode on D7. tbh i'm not sure why it even needs the resistor

dok-net, 2020-06-11:
OK, let's see. You'd have to edit `SoftwareSerial.cpp`, in `Arduino\hardware\esp8266com\esp8266\libraries\SoftwareSerial\src\SoftwareSerial.cpp`, and replace

`pinMode(m_rxPin, INPUT_PULLUP);`
by
`pinMode(m_rxPin, INPUT);`

It occurs twice.

notsolowki, 2020-06-11:
I had not looked into the library src itself but that does seem to make sense now. With your revision i may not even need the resistor

dok-net, 2020-06-11:
I've my doubts w.r.t. the sensors electrical spec, but I may be just me. I would have expected the HW UART after the `swap()` to force defined pull-up or pull-down or floating characteristics on the pins, no matter what. You could try to set the pin mode in `setup()` like above before doing `Serial.swap()` and see what happens.

dok-net, 2020-06-11:
> With your revision i may not even need the resistor

That's the point of it. I am just wary that I had it `INPUT_PULLUP` for a good reason..... :-)

dok-net, 2020-06-11:
Like, all the devices I tested so far, seem to do better with pullup, because this avoids random data when there's nothing connected, or for external devices that are just high-resistance, no pullup or down, during idle states, like on startup or power-saving. Darn.

notsolowki, 2020-06-11:
pinMode(13,INPUT); does indeed work without the resistor. Without pinMode(13,INPUT_PULLUP); 'sometimes' a reading make it through but it is very very slow.

dok-net, 2020-06-11:
And what about the diode on D8, the ESP not booting without makes no sense to me. What if you boot, and attach the sensor afterward?

notsolowki, 2020-06-11:
Hmm. i have confused myself a little bit. Now everything seems to work without any additional hardware as long as i use  pinMode(13,INPUT);. I don't see that being the problem why i couldn't upload tho. I does indeed upload just fine without any additional hardware configurations

dok-net, 2020-06-11:
OK, no problem, this way it's all making sense. So, you need to turn off PULLUP even when using HW UART via `Serial.swap()`, am I getting that right?

dok-net, 2020-06-11:
you said the "welcome" message never showed on your PC screen, that would not have been related to any receive or transmit with the sensor.

notsolowki, 2020-06-11:
With the sketch you revised for me it does not seem to work either way with INPUT or INPUT_PULLUP. however with either pinMode i see the welcome message with Serial.swap()

dok-net, 2020-06-11:
um, what?

notsolowki, 2020-06-11:
> um, what?

Maybe i have this wired wrong, in the```
 below sketch i should have sensor wire to D7 & D8?

```
/***************************************************
  Infrared CO2 Sensor 0-50000ppm(Wide Range)
* ****************************************************
  The follow example is used to detect CO2 concentration.

  @author lg.gang(lg.gang@qq.com)
  @version  V1.0
  @date  2016-6-6

  GNU Lesser General Public License.
  See <http://www.gnu.org/licenses/> for details.
  All above must be included in any redistribution
* ****************************************************/
#include <SoftwareSerial.h>
SoftwareSerial mySerial;
unsigned char hexdata[9] = {0xFF, 0x01, 0x86, 0x00, 0x00, 0x00, 0x00, 0x00, 0x79}; //Read the gas density command /Don't change the order
void setup() {
  Serial.begin(9600);
  Serial.swap();
pinMode(13,INPUT);
  while (!Serial) {
  }
  mySerial.begin(9600, SAWSERIAL_8N1, -1, 1);
  mySerial.println("Detecting CO2 concentration");
}

void loop() {
  Serial.write(hexdata, 9);
  delay(500);

  for (int i = 0, j = 0; i < 9; i++)
  {
    if (Serial.available() > 0)
    {
      long hi, lo, CO2;
      int ch = Serial.read();

      if (i == 2) {
        hi = ch;    //High concentration
      }
      if (i == 3) {
        lo = ch;    //Low concentration
      }
      if (i == 8) {
        CO2 = hi * 256 + lo; //CO2 concentration
        mySerial.print("CO2 concentration: ");
        mySerial.print(CO2);
        mySerial.println("ppm");
      }

    }

  }

}
```

dok-net, 2020-06-11:
that's a mess. You're mixing Serial and mySerial.

dok-net, 2020-06-11:
Sorry, may be my bad. As I said, not tried, not compiled :-)

notsolowki, 2020-06-11:
> Sorry, may be my bad. As I said, not tried, not compiled :-)

lol that's okay i think you were just trying to make the setu() message print anyways. Should i make everything in the loop mySerial?

dok-net, 2020-06-11:
No, everything fine in the sketch I posted for you earlier today, the one showing you the use of `Serial.swap()`. It's getting late...

dok-net, 2020-06-11:
let me delete these embarrassing messages.

dok-net, 2020-06-11:
So, I've redacted as much as I think is good for someone looking here to find information on how to fix their problem, and not get completely confused. All I would like you to clarify now, is how it makes any sense to place a diode backward between D8's TX to the sensor's RX, this is like there's a pullup in the sensor's RX?

notsolowki, 2020-06-11:
Without Serial.swap() I need no extra resistors or diodes and it works fine with pinMode(13,INUPT); I could not seem to get the sensor to work with serial swap in the earlier sketch

skupu, 2022-03-31:
> This is the code i use for the arduino UNO.
> 

Hi,
https://forum.arduino.cc/t/gravity-uart-ir-co2-sensor-sample-code/666307/8?u=skupu
Here you posted a code, which looks nice, but I need some support with it.
Could you share the page/image/index *.h files somewhere?

Thank you!

notsolowki, 2022-03-31:
@skupu 

Hopefully this helps.

`namespace index_html
{
CreatePageName;  // Include this on every of your namespaced-webpages
HTMLPAGE = R"===(
            <!DOCTYPE html PUBLIC "-//W3C//DTD HTML 4.01 Transitional//EN">
  <!DOCTYPE HTML>
  <html>
  //<br>[image]
  <body style="background-color:#000000;font-family: Arial, Helvetica, Sans-Serif; Color: #FF0000; ">
  <head>
  <meta content="text/html; charset=ISO-8859-1"
  http-equiv="content-type">
  <meta name = "viewport" content = "width = device-width, initial-scale = 1.0, maximum-scale = 1.0, user-scalable=0">
  <title>LCS Wifi</title>
  <style>
  </style>
  </head>
  <body>
  <h1>Wifi Setup</h1>
  <form action="/" method="post">  
  <label>ssid:&nbsp;</label>
  <input maxlength="30" name="ssid"><br>
  <label>Password:&nbsp;</label><input maxlength="30" name="Password"><br>
  <INPUT type="submit" value="Send"> <INPUT type="reset">
  <a href="status">Status</a>

  </P>
  </FORM>
  
  </body>
  </html>
    )===";
}`

notsolowki, 2022-03-31:
@skupu 

Hopefully this helps,

```
 namespace index_html
{
CreatePageName;  // Include this on every of your namespaced-webpages
HTMLPAGE = R"===(
            <!DOCTYPE html PUBLIC "-//W3C//DTD HTML 4.01 Transitional//EN">
  <!DOCTYPE HTML>
  <html>
  //<br><img src="/images/image.jpg">
  <body style="background-color:#000000;font-family: Arial, Helvetica, Sans-Serif; Color: #FF0000; ">
  <head>
  <meta content="text/html; charset=ISO-8859-1"
  http-equiv="content-type">
  <meta name = "viewport" content = "width = device-width, initial-scale = 1.0, maximum-scale = 1.0, user-scalable=0">
  <title>LCS Wifi</title>
  <style>
  </style>
  </head>
  <body>
  <h1>Wifi Setup</h1>
  <form action="/" method="post">  
  <label>ssid:&nbsp;</label>
  <input maxlength="30" name="ssid"><br>
  <label>Password:&nbsp;</label><input maxlength="30" name="Password"><br>
  <INPUT type="submit" value="Send"> <INPUT type="reset">
  <a href="status">Status</a>

  </P>
  </FORM>
  
  </body>
  </html>
    )===";
} 
```

[6 more replies]