plerup/espsoftwareserial #170: Odd and Even Parity exchanged

CloseToTheEdge, 2020-06-17:
Test Version: 6.8.3, 6.7.0
Test Parameter: 
SWSERIAL_7**O**2 == 7 Bit **Even** Parity
SWSERIAL_7**E**2 == 7 Bit **Odd** Parity

dok-net, 2020-06-17:
If you want this to be looked into, one may ask a little more diligence in reporting from you. I am rather convinced that I tested quite a few more data/parity/stop combinations between EspSoftwareSerial and ESP8266 HW serial.

dok-net, 2020-06-21:
Not for SW serial RX. Judging by the sloppy report, and non-response to my question, I am not inclined to spend more time on this. If you could prove that the HW UART has odd/even wrong on TX, fine. Thanks for apparently wasting my time...

dok-net, 2020-06-22:
Dear @CloseToTheEdge, I appreciate the fact that you found this bug, I just equally wish, you had provided more background, like testing more of the modes. The HW serial API in ESP8266/ESP32 disregard parity on RX, so testing against those turns out has been futile previously. I finally found some time to test in depth, and I've found the cause and I am about to fix it.

CloseToTheEdge, 2020-06-23:
Still the same Problem: Version 6.8.5

`SoftwareSerial swSer;`
`swSer.begin(1200, SWSERIAL_7O2, D2, D1, false, 256, 256); `
TX sends over D1 EVEN parity instead of ODD.

`SWSERIAL_7E2` sends ODD parity

(RX not tested)

dok-net, 2020-06-23:
Are you certain? Have you read the README, in particular the section titled `Using and updating EspSoftwareSerial in the esp8266com/esp8266 Arduino build environment`?

closed 2020-06-23 by dok-net

dok-net, 2020-06-24:
The fixed EspSoftwareSerial is now part of Arduino core for ESP8266 git master, and pending release, in version 2.7.2 of that.

commit 548e151b (2020-06-22): Parity calculation for odd data bit counts was inverted. Fixes #170