plerup/espsoftwareserial #175: after run several minutes Esp8266 will reboot and cannot connect to wifi

yumupinglan, 2020-08-19:
I see below exception, what's this mean?
```
Exception 0: Illegal instruction
PC: 0x402021e4: circular_queue ::push(unsigned int&&) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/10.1.0/bits/atomic_base.h line 420
EXCVADDR: 0x00000000

Decoding stack results
0x40100e2c: check_poison_block(umm_block*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_poison.c line 95
0x4010079c: interrupt_handler(void*, void*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.cpp line 166
0x40100cd1: umm_free_core(void*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.cpp line 351
0x401006d8: interrupt_handler(void*, void*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.cpp line 137
0x401003b0: SoftwareSerial::rxBitISR(SoftwareSerial*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/10.1.0/bits/unique_ptr.h line 420
0x401003b0: SoftwareSerial::rxBitISR(SoftwareSerial*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/10.1.0/bits/unique_ptr.h line 420
0x4010079c: interrupt_handler(void*, void*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.cpp line 166
0x401007b8: interrupt_handler(void*, void*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.cpp line 174
0x4010079c: interrupt_handler(void*, void*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.cpp line 166
0x40100e2c: check_poison_block(umm_block*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_poison.c line 95
0x401006d8: interrupt_handler(void*, void*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.cpp line 137
0x401006d8: interrupt_handler(void*, void*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.cpp line 137
0x40204784: loop_task(ETSEvent*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 208
0x401004fc: ets_post(uint8, ETSSignal, ETSParam) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 177
0x401004fc: ets_post(uint8, ETSSignal, ETSParam) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 177
0x401004fc: ets_post(uint8, ETSSignal, ETSParam) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 177
0x401004fc: ets_post(uint8, ETSSignal, ETSParam) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 177
0x401004fc: ets_post(uint8, ETSSignal, ETSParam) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 177
0x401006f0: interrupt_handler(void*, void*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.cpp line 138
0x401004fc: ets_post(uint8, ETSSignal, ETSParam) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 177
0x401003b0: SoftwareSerial::rxBitISR(SoftwareSerial*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/10.1.0/bits/unique_ptr.h line 420
0x401003b0: SoftwareSerial::rxBitISR(SoftwareSerial*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/10.1.0/bits/unique_ptr.h line 420
0x401003b0: SoftwareSerial::rxBitISR(SoftwareSerial*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/10.1.0/bits/unique_ptr.h line 420
0x4010079c: interrupt_handler(void*, void*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.cpp line 166
0x401007b8: interrupt_handler(void*, void*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.cpp line 174
0x4010079c: interrupt_handler(void*, void*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.cpp line 166
0x401004fc: ets_post(uint8, ETSSignal, ETSParam) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 177
0x401007b8: interrupt_handler(void*, void*) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.cpp line 174
0x401005a2: millis() at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring.cpp line 193
0x402021dc: std::_Function_handler   >::_M_invoke(const std::_Any_data &, unsigned int &&) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/libraries/SoftwareSerial/src/SoftwareSerial.cpp line 423
0x40201ca8: circular_queue ::for_each(Delegate  const&) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/10.1.0/bits/atomic_base.h line 122
0x401001ce: SoftwareSerial::writePeriod(unsigned int, unsigned int, bool) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/libraries/SoftwareSerial/src/SoftwareSerial.cpp line 263
0x40201f88: SoftwareSerial::rxBits() at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/10.1.0/bits/std_function.h line 303
0x401004fc: ets_post(uint8, ETSSignal, ETSParam) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 177
0x402021d0: std::_Function_handler   >::_M_invoke(const std::_Any_data &, unsigned int &&) at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/libraries/SoftwareSerial/src/SoftwareSerial.cpp line 423
0x4020483b: __yield() at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_features.h line 92
0x402015fa: PZEM004Tv30::recieve(unsigned char*, unsigned short) at /home/willieyu/Arduino/libraries/PZEM-004T-v30-master/PZEM004Tv30.cpp line 469
0x402016aa: PZEM004Tv30::sendCmd8(unsigned char, unsigned short, unsigned short, bool, unsigned short) at /home/willieyu/Arduino/libraries/PZEM-004T-v30-master/PZEM004Tv30.cpp line 246
0x40201752: PZEM004Tv30::updateValues() at /home/willieyu/Arduino/libraries/PZEM-004T-v30-master/PZEM004Tv30.cpp line 377
0x402018dd: PZEM004Tv30::voltage() at /home/willieyu/Arduino/libraries/PZEM-004T-v30-master/PZEM004Tv30.cpp line 130
0x40201077: getPzem() at /home/willieyu/Arduino/PZEMSoftwareSerial/PZEMSoftwareSerial.ino line 119
0x40201374: loop() at /home/willieyu/Arduino/PZEMSoftwareSerial/PZEMSoftwareSerial.ino line 78
0x40204928: loop_wrapper() at /home/willieyu/arduino/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 197

```

dok-net, 2020-08-27:
I may be that you have to completely erase and reflash your ESP8266. Please report if that solves it.

dok-net, 2020-09-17:
In latest Github Arduino Core for ESP8266, there's an updated EspSoftwareSerial. If you are not familiar with using that, it's all documented there. If you want to just update to the latest EspSoftwareSerial, instructions are in the README here.
Please report if that solves your issue.

dok-net, 2020-10-12:
@yumupinglan I'll be closing this in a week, unless there's new activity.

dok-net, 2020-11-19:
Closing after a month of inactivity.

closed 2020-11-19 by dok-net