plerup/espsoftwareserial #176: Hex printing fix in example onewiretest

issalig, 2020-08-19:
Should not be < 0x10 ?

https://github.com/plerup/espsoftwareserial/blob/e51872ca201d307a57ca03cbce4d18497c9daef3/examples/onewiretest/onewiretest.ino#L45

dok-net, 2020-08-27:
Could you test, report, and rephrase your question into a PR if your are right about this, please? I DO THINK that you are on to something, so please go ahead.

issalig, 2020-08-27:
I did not create a PR because it was such a small change.

Current code adds a "0" when hexadecimal is smaller than 1, thus, it will generate 00,1,2,3,...,f.

With my change < 0x10, it will add "0" to values smaller than 0x10, that is, 00,01,.., 09,  and from 10 then will be no padding

dok-net, 2020-08-27:
> I did not create a PR because it was such a small change.

That's not the way of thinking about this :-) Do you want me to do all the work for you, why is that? Please go ahead and earn your right to have your name in in the change history of this project.

commit c7e15654 (2020-08-28): Fix #176 (#178)

Small fix for padding hex values in onewiretest.ino example
closed 2020-08-28 by dok-net