plerup/espsoftwareserial #179: WDT reset software at 9600 baud rate

horaciocastro7, 2020-08-27:
Using ESP Boards 2.2.4 and earlier I get this problem. when run from a browser the
http://192.168.110.10/ppal and when executing this same procedure about 12 times there is a memory dump.

This only happens with speeds of 9600 because when doing them at 57600 or 115200 this does not happen. I know it is the SoftwareSerial library because if I run this with Serial and at the same speed, this does not happen.

I appreciate in advance any help in this regard.

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <ESP8266WebServer.h>
ESP8266WebServer server(80);

#include "SoftwareSerial.h"/
SoftwareSerial mySerial;

#define GDencabezado_prueba1 F("HOLA MUNDO")
#define GDencabezado_prueba2 F("HOLA MUNDO 2")
#define GDencabezado_prueba3 F("HOLA MUNDO 3")

#ifndef STASSID
/Datos del router/
#define GSssid_nombre ""
#define GSssid_clave ""
/Configuracion de la IP/
byte ip[4] = { 192, 168, 110, 10 };
byte gateway[4] = { 192, 168, 110, 1 };
byte subnet[4] = { 255, 255, 255, 0 };
#endif

void setup() {
wdt_disable();
Serial.println(115200);
mySerial.begin(9600, SWSERIAL_8N1, -1, D6, false, 128, 0);
mySerial.enableIntTx(false);

WiFi.setSleepMode(WIFI_NONE_SLEEP); WiFi.mode(WIFI_STA);
WiFi.config(ip, gateway, subnet);
WiFi.begin(GSssid_nombre, GSssid_clave);
// Esperar a que nos conectemos
byte zcontador = 0;
while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    delay(200);
    zcontador++;
    if (zcontador >= 40) {
        ESP.restart();
    }
}

server.on(F("/ppal"), HTTP_GET, []() { 
 server.send(200, F("text/plain"), "OK");
 mySerial.println(GDencabezado_prueba1); yield();
 mySerial.println(GDencabezado_prueba2); yield();
 mySerial.println(GDencabezado_prueba3); yield();

 Serial.println(GDencabezado_prueba1); yield();
    });
server.begin();
wdt_reset();
ESP.wdtEnable(WDTO_8S);
}

// Add the main program code into the continuous loop() function
unsigned long int GLtiempo = millis();
int zcontador = 0;
void loop() {
if (millis() - GLtiempo >= 30000) {
GLtiempo = millis();
Serial.print(String(millis() / 1000) + '-');
zcontador++;
if (zcontador == 20) {
[... 289 lines cut ...]
3ffffef0: 00000000 3fffdad0 3ffee5c8 00000030
3fffff00: 00000000 3fffdad0 3ffee5c8 00000030
3fffff10: 00000000 4bc6a7f0 21cac083 4020531a
3fffff20: 00000000 00000000 4bc6a7f0 00000000
3fffff30: 00000000 00000001 3fffff50 402095dd
3fffff40: 3fffdad0 00000000 3ffee440 402034d9
3fffff50: 40209938 00000000 00001388 40208039
3fffff60: 007a1200 483efc0b 3ffee700 3ffee5c8
3fffff70: 00000000 00000000 3ffee3dc 40203644
3fffff80: 00000000 00000000 00000001 40100548
3fffff90: 3fffdad0 00000000 3ffee588 40100569
3fffffa0: 3fffdad0 00000000 3ffee588 40206079
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

ets Jan 8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
VMDPR_115200

dok-net, 2020-08-28:
Could you please a the stack decoder output as well (again ;-) )?

devyte, 2020-08-28:
@horaciocastro7 said:
> ESP Boards 2.2.4 and earlier

2.2.4? really?
- If that's the core version you're using, the latest release is 2.7.4. No support is provided for such ancient core versions. Core issues are accepted for latest release and later (latest git), and a similar policy is followed in this repo, because specific versions of this lib are tied to core releases.
- If that is the library version for software serial, the latest release is 6.8.5. I doubt such an old version will be looked at here.
- Is that a typo and you meant 2.7.4?

Please clarify.

devyte, 2020-08-28:
Also, in setup() you're calling:
`wdt_disable()`
which is defined to be:
`EspClass::wdtDisable()`
which in turn calls Espressif's api:
`system_soft_wdt_stop()`

Then, at the end of setup() you're calling:
`wdt_reset()`
which is defined to be:
`EspClass::wdtFeed()`
which in turn calls Espressif's api:
`system_soft_wdt_feed()`

However, [Espressif's 2c api reference](https://www.espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf) section 3.3.37 on system_soft_wdt_feed() says:
> This API can only be called if software watchdog is enabled

and only after that are you calling `ESP.wdtEnable(WDTO_8S)`

Given that your code disables the watchdog, but then feeds it, and that it hits a sw wdt, I'd say your usage isn't correct.

horaciocastro7, 2020-08-29:
> @horaciocastro7 said:
>
>
> 2.2.4? really?
>
> * If that's the core version you're using, the latest release is 2.7.4. No support is provided for such ancient core versions. Core issues are accepted for [...]

If it is version 2.7.4 I was wrong in the wording.

horaciocastro7, 2020-08-29:
> Also, in setup() you're calling:
> `wdt_disable()`
> which is defined to be:
> `EspClass::wdtDisable()`
> which in turn calls Espressif's api:
> `system_soft_wdt_stop()`
>
> Then, at the end of setup() you're [...]

void setup() {
	Serial.println(115200);
	mySerial.begin(9600, SWSERIAL_8N1, -1, D6, false, 128, 0);
	mySerial.enableIntTx(false);

    WiFi.setSleepMode(WIFI_NONE_SLEEP); WiFi.mode(WIFI_STA);
    WiFi.config(ip, gateway, subnet);
    WiFi.begin(GSssid_nombre, GSssid_clave);
    // Esperar a que nos conectemos
    byte zcontador = 0;
    while (WiFi.waitForConnectResult() != WL_CONNECTED) {
        delay(200);
        zcontador++;
        if (zcontador >= 40) {
            ESP.restart();
        }
    }

    server.on(F("/ppal"), HTTP_GET, []() { 
     server.send(200, F("text/plain"), "OK");
     mySerial.println(GDencabezado_prueba1); yield();
     mySerial.println(GDencabezado_prueba2); yield();
     mySerial.println(GDencabezado_prueba3); yield();

     Serial.println(GDencabezado_prueba1); yield();
        });
    server.begin();
    ESP.wdtEnable(WDTO_8S);
    wdt_reset();
}

The blockade is presented again

Opening port
Port open
30-HOLA MUNDO
60-HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: sys
sp: 3fffed40 end: 3fffffb0 offset: 01a0
3fffeee0:  00000018 00012e0d 00000000 00000000  
3fffeef0:  4021f922 3ffec548 3ffec548 3ffe93bc  
3fffef00:  00000000 3ffece38 00000000 00000000  
3fffef10:  4022d208 3ffece38 3ffeddf0 3ffec548  
3fffef20:  4022fb75 3fffdcc0 3ffe8b20 3ffe8b20  
3fffef30:  4022fbba 3fffdab0 00000000 3fffdcb0  
3fffef40:  3ffede08 3fffdad0 3ffee5c8 40205e5c  
3fffef50:  40000f49 40000f49 3fffdab0 40000f49  
3fffef60:  40000e19 00000005 000468cc 00000000  
3fffef70:  00000000 aa55aa55 000000d5 40104f3d  
[... 281 lines cut ...]
0x4020f632: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x4010012d: SoftwareSerial::preciseDelay(bool) at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\SoftwareSerial\src\SoftwareSerial.cpp line 235
0x401001e6: SoftwareSerial::writePeriod(unsigned int, unsigned int, bool) at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\SoftwareSerial\src\SoftwareSerial.cpp line 263
0x40210380: ethernet_output at netif/ethernet.c line 312
0x40216714: etharp_output_to_arp_index at core/ipv4/etharp.c line 770
0x4010012d: SoftwareSerial::preciseDelay(bool) at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\SoftwareSerial\src\SoftwareSerial.cpp line 235
0x40100548: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 177
0x40205f1e: __yield() at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 112
0x40205f58: optimistic_yield(uint32_t) at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 151
0x40100548: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 177
0x40100548: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 177
0x40100548: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 177
0x40100548: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 177
0x40204db6: WiFiClient::WiFiClient() at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src\WiFiClient.cpp line 81
0x4020531c: WiFiServer::available(unsigned char*) at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src\WiFiServer.cpp line 141
0x40205fec: loop_wrapper() at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 192
0x40205fec: loop_wrapper() at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 192
0x40205316: WiFiServer::available(unsigned char*) at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src\WiFiServer.cpp line 140
0x40209569: WiFiClient::operator bool() at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src\WiFiClient.cpp line 349
0x402034d5: esp8266webserver::ESP8266WebServerTemplate ::handleClient() at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h line 312
0x40207fc5: String::invalidate() at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\WString.cpp line 140
[... cut ...]
0x40100548: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 177
0x40100569: esp_schedule() at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 125
0x40206009: loop_wrapper() at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 199

dok-net, 2020-08-29:
@horaciocastro7 If you can manage to checkout from git as per the README instructions of the ESP8266 Core for Arduino project, and pull the very latest commit of the there-in contained EspSoftwareSerial master branch (submodule in ./libraries/SoftwareSerial), I'd ask you to rerun your sketch and report if that fixes your issue.

horaciocastro7, 2020-08-30:
today update all boards by first removing Arduino15/staging/packages -> esp8266-2.7.4
Then removing boards 2.7.4 and downloading today's github and uploading it to boards and the continuing issue takes a little longer to block but ends up doing it.

Opening port
Port open
30-HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
60-HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO
HOLA MUNDO

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: sys
sp: 3fffed40 end: 3fffffb0 offset: 01a0
3fffeee0:  00000018 00012e0d 00000000 3ffe88f4  
3fffeef0:  4010559d 017e4782 60000600 60000600  
3fffef00:  00000000 3ffece38 00000000 00000000  
3fffef10:  4022d208 3ffece38 3ffeddf0 60000600  
3fffef20:  4022fb75 3ffece38 3ffeddf0 04c7b71b  
3fffef30:  4022fbba 3fffdab0 00000000 3fffdcb0  
3fffef40:  3ffede18 3fffdad0 3ffee5c8 40205e5c  
3fffef50:  40000f49 40000f49 3fffdab0 40000f49  

In exception decoder I get this result

Decoding stack results
0x4022d208: credito_venta(unsigned char, String) at C:\xampp\htdocs\I7\.. Modulo y Pantalla I7\I7/DCrEf.ino line 574
0x4022fb75: touch_recepcion() at C:\xampp\htdocs\I7\.. Modulo y Pantalla I7\I7/PNeEr.ino line 329
0x4022fbba: touch_recepcion() at C:\xampp\htdocs\I7\.. Modulo y Pantalla I7\I7/PNeEr.ino line 335
0x40205e5c: precio_disp_organizar(unsigned char) at C:\xampp\htdocs\I7\.. Modulo y Pantalla I7\I7/QGPrecio.ino line 379
0x40100548: Twi::onTwipEvent(unsigned char) at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_si2c.cpp line 565
0x4020f568: printer_tiempo_entrerecibos() at C:\xampp\htdocs\I7\.. Modulo y Pantalla I7\I7/CRecPrin.ino line 711
0x40100548: Twi::onTwipEvent(unsigned char) at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_si2c.cpp line 565
0x4020fcd9: eeprom2_writeBytes(int, int, unsigned char*) at C:\xampp\htdocs\I7\.. Modulo y Pantalla I7\I7/PMem.ino line 376
0x4020f632: protocolo_dinero_actual(unsigned char) at C:\xampp\htdocs\I7\.. Modulo y Pantalla I7\I7/QGilbarco.ino line 498
[... 1 lines cut ...]
0x40229bca: touch_isleros(String) at C:\xampp\htdocs\I7\.. Modulo y Pantalla I7\I7/PNeEr.ino line 1008
0x40100548: Twi::onTwipEvent(unsigned char) at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_si2c.cpp line 565
0x40228c87: touch_precio_cambio(String) at C:\xampp\htdocs\I7\.. Modulo y Pantalla I7\I7/PNeEr.ino line 718
0x40100548: Twi::onTwipEvent(unsigned char) at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_si2c.cpp line 565
0x4010012d: inso_serial_331::rxRead() at C:\Users\Inso Blanco\Documents\Arduino\libraries\inso_serial_331\inso_serial_331.cpp line 205
0x401001e6: sws_isr_0() at C:\Users\Inso Blanco\Documents\Arduino\libraries\inso_serial_331\inso_serial_331.cpp line 38
0x40210380: eeprom2_writeByte(int, char) at C:\xampp\htdocs\I7\.. Modulo y Pantalla I7\I7/PMem.ino line 555
0x40216714: touch_perifericos_actualizar() at C:\xampp\htdocs\I7\.. Modulo y Pantalla I7\I7/PNeEr.ino line 1740
0x40100548: Twi::onTwipEvent(unsigned char) at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_si2c.cpp line 565
0x40100548: Twi::onTwipEvent(unsigned char) at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_si2c.cpp line 565
0x40100548: Twi::onTwipEvent(unsigned char) at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_si2c.cpp line 565
0x40100548: Twi::onTwipEvent(unsigned char) at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_si2c.cpp line 565
0x40100548: Twi::onTwipEvent(unsigned char) at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_si2c.cpp line 565
0x40100548: Twi::onTwipEvent(unsigned char) at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_si2c.cpp line 565
0x40205f3d: precio_disp_dato_tx0_grade(unsigned char) at C:\xampp\htdocs\I7\.. Modulo y Pantalla I7\I7/QGPrecio.ino line 431
0x40205316: rtc_ram_lectura_prueba() at C:\xampp\htdocs\I7\.. Modulo y Pantalla I7\I7/PCalRel.ino line 893
0x40100548: Twi::onTwipEvent(unsigned char) at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_si2c.cpp line 565
0x40100548: Twi::onTwipEvent(unsigned char) at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_si2c.cpp line 565
0x40205316: rtc_ram_lectura_prueba() at C:\xampp\htdocs\I7\.. Modulo y Pantalla I7\I7/PCalRel.ino line 893
0x40209569: fs_inicio() at C:\xampp\htdocs\I7\.. Modulo y Pantalla I7\I7/AFs.ino line 123
0x402034d5: combustible_manguera_id(unsigned char, unsigned char) at C:\xampp\htdocs\I7\.. Modulo y Pantalla I7\I7/CPrecVen.ino line 964
[... cut ...]
0x40100548: Twi::onTwipEvent(unsigned char) at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_si2c.cpp line 565
0x40100569: Twi::onTwipEvent(unsigned char) at C:\Users\Inso Blanco\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_si2c.cpp line 576
0x40206009: precio_disp_organizar_grade(unsigned char, unsigned char, String) at C:\xampp\htdocs\I7\.. Modulo y Pantalla I7\I7/QGPrecio.ino line 353

dok-net, 2020-08-30:
Have you noticed how you stack decode doesn't match the symbols from the sketch you supposedly were running?
Also, please don't `mySerial.enableIntTx(false);`!

dok-net, 2020-08-30:
Oh great, you edited your posting.

dok-net, 2020-08-30:
How would you explain that it never prints GDencabezado_prueba2 or 3 ???

horaciocastro7, 2020-08-30:
Serial.println (GDheader_test1);  yield () what I show is the internal Serial of esp8266 to be able to do the debugging

horaciocastro7, 2020-08-30:
I made the modification that you told me and it worked, it no longer hangs.

	//mySerial.enableIntTx(false);

I made the modification that you told me and it worked no longer it is blocked by the guard dog.

Thank you very much for your help.

//mySerial.enableIntTx(false);
What is function used for and in which cases should this function be used or removed?

Thank you again for your help

horaciocastro7, 2020-08-30:
In another program when I placed mySerial.enableIntTx(true) it randomly presented small problems of rare characters in the transmission very few but I had working them at 57600 and when I placed mySerial.enableIntTx(false) these random errors very occasionally disappeared. so I decided to use it better.

closed 2020-08-30 by horaciocastro7