plerup/espsoftwareserial #180: What Am I doing Wrong? I had this ESP8266 Software serial working months ago but now.... NOPE

WheresWally, 2020-09-12:
This is the simplest test I could make. A brand new ESP8266 NodeMCU and a wire joining D6-D7.
The WiFi is off. I just cant think what else to do.
I even wrote code to test the pins by making them HIGH/LOW and that was all good.

`#include <SoftwareSerial.h>
#define FONA_RX D6
#define FONA_TX D7
SoftwareSerial fonaSS;

void TestComPort()
{    Serial.print("\nTEST COM"); delay(1999);
char c;
int i=0;
 do {
    Serial.print("\nTX=>");      delay(1000);
    fonaSS.println(millis());    delay(1000);
    Serial.print(millis());
     i++;
    Serial.print(" RX=>");     delay(1000);
    while( fonaSS.available() > 0) {
       c=fonaSS.read();         delay(100);
       Serial.print(c);         delay(100);
         }  
    } while(i<100);
}

void setup() {
  Serial.begin(115200);   //Initialize hardware serial with baudrate of 115200
  fonaSS.begin(115200,SWSERIAL_8N1,FONA_RX, FONA_TX, false,95,11); //Define hardware connections

  Serial.println("\nSoftware serial test started");
}

void loop() {
TestComPort();

//  while (fonaSS.available() > 0) {    Serial.write(fonaSS.read());    yield();  }
//  while (Serial.available() > 0) {fonaSS.write(Serial.read());  yield();  }

}`

WheresWally, 2020-09-12:
OK, if I drop the baud to 57600 it works.
`  fonaSS.begin(57600,SWSERIAL_8N1,FONA_RX, FONA_TX, false,64); //Define hardware connections
`
Hmmmm?

dok-net, 2020-09-12:
I'm asking for due diligence in these reports. Please review your original line
`fonaSS.begin(115200,SWSERIAL_8N1,FONA_RX, FONA_TX, false,95,11);`
and your latter
`fonaSS.begin(57600,SWSERIAL_8N1,FONA_RX, FONA_TX, false,64);`.
Obviously, your claim `drop the baud to 57600`isn' the whole story here.

Please think about why you specified the ISR buffer size as 11 in your first attempt, and if that is a mistake on your side. IIRC correctly, the line of code you copied from the examples, is accompanied by a sufficiently explanatory comment, but YMMV and I might have to make a larger warning sign, so to speak.

What do you think?

WheresWally, 2020-09-12:
To be honest I have little idea what the 95,11 is setting, I copied that from the example code in swsertest.ino
I was not sure so I set it to 64, and let the default do the other.
HOWEVER I am finding that a baud of 57600 works where as 115200 does not.
I would love to know what has changed as I was happily reading the serial output at 115200 from a SIM5320E 4G GMS/GPS board for years now.
So can SoftwareSerial run happily at 115200 on an ESP8266-12E?

WheresWally, 2020-09-12:
One I had the line of code below working, I tried the next lines of code.
`fonaSS.begin(57600,SWSERIAL_8N1,FONA_RX, FONA_TX, false,64);`
Next
`fonaSS.begin(115200,SWSERIAL_8N1,FONA_RX, FONA_TX, false,64);`
and then
`fonaSS.begin(115200,SWSERIAL_8N1,FONA_RX, FONA_TX, false,256);`
and then
`fonaSS.begin(57600,SWSERIAL_8N1,FONA_RX, FONA_TX, false,256);`
The 115200 baud fails, the 57600 works.
Sorry, I get that this is annoying basic stuff for you.

dok-net, 2020-09-12:
Let's look at swsertest.ino, which is the example you presumably uses as the template for your code.
It says right at the top:
```
// On ESP8266:
// At 80MHz runs up 57600ps, and at 160MHz CPU frequency up to 115200bps with only negligible errors.
```
And further down:
```
// Reminder: the buffer size optimizations here, in particular the isrBufSize that only accommodates
// a single 8N1 word, are on the basis that any char written to the loopback SoftwareSerial adapter gets read
// before another write is performed. Block writes with a size greater than 1 would usually fail.
SoftwareSerial swSer;

void setup() {
        Serial.begin(115200);
        swSer.begin(BAUD_RATE, SWSERIAL_8N1, D5, D6, false, 95, 11);
```
Now, you may say that you didn't know what `isrBufSize` is - but on the other hand, this can be taken as a warning to clearly understand what you are doing, and read the docs and headers before just cloning any line of source code.

In order to avoid problems as you are met with from repeated occuring, may I ask you to help and redesign the comments such that you would have done that extra research first?

As for your code in this issue, I would like to point out that 115200bps might just work fine after you've enlarged the ISR buffer size, as long as you understand the requirements and implications, for your actual use. Local pin-to-pin loopback is an extreme real-time full duplex case, which may not be relevant to your external board at all. Just think about it, SW serial cannot send and receive at the same time, since there is no asynchronous HW buffering possible.

WheresWally, 2020-09-13:
ahhh. ok derps aplenty on my end.
I shall try again with the fono board.

The only question is why did the code work 12 months ago, and now that I am recompiling with the newer versions of Arduino IDE and ESP8266, the code to do serial comms to the GMS/GPS module do not work at 115200. 

My code had to be modified for the new style  of initialization so I wondered what else might have changed in the SoftwareSerial code base.

WheresWally, 2020-09-13:
Thanks so much for your help. Can I buy you a coffee or a beer?

WheresWally, 2020-09-13:
Feel free to close this now that it is my derp entirely.

closed 2020-09-17 by dok-net