plerup/espsoftwareserial #182: On ESP32 can also use GPIO36 and GPIO39 to recieve, but need to edit "SoftwareSerial.cpp"

Rob58329, 2020-11-15:
re. plerup/espsoftwareserial

On ESP32 one can also use GPIO36 and GPIO39 to recieve, but need to edit "SoftwareSerial.cpp":

Specifically, if one changes the code "(pin >= 32 && pin <= 35);" to "(pin >= 32 && pin <= 39);" in the above file, then GPIO36 and GPIO39 will work fine as Rx pins (with external pullups if necessary).

Suggest you consider updating "SoftwareSerial.cpp" to reflect this?
(PS. Neat software!)

Regards, Rob (15Nov20)

closed 2020-11-15 by dok-net

dok-net, 2020-11-15:
@Rob58329 I'm sorry this isn't quite what you requested, but thanks for giving me an opportunity to look into this anew. If you check out the documentation (link is in commit message), I hope you will agree that for general purpose use and least surprise, let alone needing to find out what happens when using the `INPUT_PULL` flag on GPIOs that don't support this, is the right way to go.

Rob58329, 2020-12-29:
@dok-net - Thank you for considering my suggestion!

(1) I note that you have changed the line to "<39".  As my "NodeMCU-32S" ESP32 board has a pin GPIO39, I suggest that this "<39" should be "<=39"??

(2) For you info, one of my ESP32 projects successfully uses 6 serial ports as follows:

```
Serial.begin (115200,SERIAL_8N1, 35,13); // (HardwareSerial) // rx_pin,tx_pin
Serial1.begin(115200,SERIAL_8N1, 34,33); // (HardwareSerial) // rx_pin,tx_pin
Serial2.begin(115200,SERIAL_8N1, 39,25); // (HardwareSerial) // rx_pin,tx_pin
Serial3s.begin(9600,SWSERIAL_8N1, 4,16,false,95,200); // (SoftwareSerial) // rx_pin,tx_pin
Serial4s.begin(9600,SWSERIAL_8N1,22,23,false,95,200); // (SoftwareSerial) // rx_pin,tx_pin
Serial5s.begin(9600,SWSERIAL_8N1,36,19,false,95,200); // (SoftwareSerial) // rx_pin,tx_pin // NB. need to edit the SoftwareSerial library to allow GPIO36 use
```

As you commented, you have now added the lines "#elif defined(ESP32)" and "&& (pin < 34)" into your espsoftwareserial code which stops the use of GPIO 34 to 39.

This surprises me as (1) The standard HardwareSerial ports all allow use of pins 34 to 39, (2) your espsoftwareserial code works FINE using ESP32 pins 34 to 39 as long as I comment out your above added 2 lines.

(As discussed, these ESP32 pins 34 to 39 only work as Receive Pins, and don't have internal pullups so may need external pullups depending on what you connect them to.  However this is well documented, and is exactly the same if using Hardware Serial or Software Serial, so I don't feel this is a reason to disable their use in espsoftwareserial.)  (PS. it appears from my tests that setting the INPUT_PULL flag on these pins simply does not do anything.)

Therefore you may wish to consider removing your above GPIO 34 to 39 restriction?

PS.  I find your "plerup/espsoftwareserial" very useful & reliable - thank you!!

commit d5060653 (2020-11-15): According to https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/gpio.html,

GPIOs from 36 to 39 can be utilized for output on ESP32-S2, but there are issues with IRQ and internal pullups. Hence, GPIOs 34 and 35 must
be set to blocked for Rx to avoid conflicting use of the input w/ pullup mode that's used.
Inspired by, half-way fixes #182