plerup/espsoftwareserial #183: New in version 6.10.0: "::availableForWrite() marked override, but does not override"

ba58smith, 2020-11-18:
When trying to build a PlatformIO Project that uses EspSoftwareSerial, I get `.pio/libdeps/d1_mini/EspSoftwareSerial/src/SoftwareSerial.h:132:9: error: 'int SoftwareSerial::availableForWrite()' marked override, but does not override`, but only for my ESP8266 Project. My ESP32 project builds correctly.

When I revert to version 6.9.0, PlatformIO for some reason uses version 6.8.5, but the Project builds correctly for ESP8266 and ESP32, with no other changes.

Everything was working until version 6.10.0 was released a few days ago.

dok-net, 2020-11-19:
You're using ESP32 as an object identifier, whereas ESP32 is a `#define` :-)

dok-net, 2020-11-19:
Your's is a recurring question on each new release :-) Read the fine README.
TL;DR: Don't use the EspSoftwareSerial from the Arduino library repository with the ESP8266, it's already versioned and part of the ESP8266 Arduino core. For ESP32, do use the Arduino library repository, though.

closed 2020-11-19 by dok-net