plerup/espsoftwareserial #185: size_t readBytes(uint8_t* buffer, size_t size) override; ESP32

kang2k10, 2020-11-30:
Arduino ide 1.8.13
ESP32 core 1.04
Espsoftwareserial 6.10.0
Compiling code error

C:\Users\owner\Documents\Arduino\libraries\EspSoftwareSerial\src/SoftwareSerial.h:166:12: error: 'size_t SoftwareSerial::readBytes(uint8_t*, size_t)' marked 'override', but does not override
     size_t readBytes(uint8_t* buffer, size_t size) override;
            ^
C:\Users\owner\Documents\Arduino\libraries\EspSoftwareSerial\src/SoftwareSerial.h:169:12: error: 'size_t SoftwareSerial::readBytes(char*, size_t)' marked 'override', but does not override
     size_t readBytes(char* buffer, size_t size) override {

dok-net, 2020-11-30:
This is a recurring question. Arduino library dependency resolution problems. Should be somewhere in the README or previous, closed, issues here. Could you please search for that, and add a reference here? If there's really nothing like that for ESP32, just ESP8266, it deserves to be added, I'd take that on myself.

kang2k10, 2020-11-30:
ok. i modify library code, use from https://github.com/plerup/espsoftwareserial/pull/109/files#diff-0aa8881a1822639fa0bdddebab500c293ae8b7a95562dcac97b6c6f4c45dd0e1

closed 2020-12-29 by dok-net