plerup/espsoftwareserial #186: One wire example doesn't work

dduehren, 2020-12-03:
I have an 8266 connected to nothing else.  Wire between GPIOs 12 and 14, running onewire.ino example.  No characters are received.  Don't know where to start to debug this.   I'm using a Node12MCU.  Any suggestions.
Am interested in getting this to work with modified software serial for Arduino.
https://github.com/nickstedman/SoftwareSerialWithHalfDuplex/issues/2
I have that software running between two UNOs just fine.  I may try that library on 8266.

dduehren, 2020-12-03:
Have it working in one direction with 8266 as sender, Uno as receiver.  Thus it appears that the problem is on the receive side.

dok-net, 2021-01-25:
The test isn't exactly sane, but I am sure you are not using it in the right way, either. I quote: "8266 as sender, Uno as receiver". For a test that toggles between tx and rx on each line, therefore obviously relying on sequential simplex communication, what's the meaning of as sender, as receiver.
I'm sure you'll figure this out. Not an issue with EspSoftwareSerial.

closed 2021-01-25 by dok-net