plerup/espsoftwareserial #189: int availableForWrite() override

manudax, 2020-12-21:
Hello,
since the 6.10.0 version Arduino generate an error :
"SoftwareSerial.h:132:9: error: 'int SoftwareSerial::availableForWrite()' marked override, but does not override"

Arduino V 1.8.13
with a project including PZEM004T.h and ESP8266 platform

Workaround ; use the espsoftwareserial V 6.9.0

Manudax.

dok-net, 2020-12-29:
Please always mention whether you are compiling for ESP8266 or ESP32. And remember to read the documentation, it clearly states the dependency between versions of EspSoftwareSerial and the Arduino core for your platform.

closed 2020-12-29 by dok-net

manudax, 2020-12-29:
Hello Mr Dok-Net, as mentioned in my first message I use PZEM004T.h and ESP8266 platform the compilation is for ESP8266. I compare two versions one runnig and the last not running. I know what changed as mentioned "SoftwareSerial.h". The code changed and now "SoftwareSerial.h" clearly does not support any more the PZEM004T with the last version. The last version of Arduino does not compile the PZEM004T because "SoftwareSerial.h" included in this version changed.

dok-net, 2020-12-30:
@manudax Please don't dwell on my generic reply, but instead, do the documentation which clearly instructs you how to use EspSoftwareSerial with ESP8266.
TL;DR? Don't install EspSoftwareSerial from Arduino library manager!