plerup/espsoftwareserial #191: GPIO35 blocked for RX

automatromsrl, 2020-12-26:
I'm using GPIO35 for RX (9bit 9600bps) on few projects and since 6.10.0 this pin cannot be used for Software Serial RX. Can I manually modify the library to enable RX on GPIO35 considering that I'm not using IRQ nor internal pull-up for GPIO34-39?
Using 6.9.0 did not rised me any troubles with the boards in field until now.

Kind regards,
Mihai

dok-net, 2020-12-29:
For ease of use and to prevent unexpected side effects, the banned GPIOs were chosen carefully, the lack of internal pullup being the reason here. You're free to patch you local copy, obviously.

dok-net, 2020-12-30:
@automatromsrl I've decided to see things your way after all :-) Could you please pull and test PR #192 ?

closed 2020-12-31 by dok-net