plerup/espsoftwareserial #194: override

Paedda801, 2021-01-19:
Hi,
why do you use override in EspSoftwareSerial.h

#if defined(ESP8266)
    int availableForWrite() override {
#else
    int availableForWrite() {
#endif

In the base-class of SoftwareSerial (Stream) there is now function availableForWrite() and my Arduino-IDE (ESP8266-Package, 2.7.4) doesn't like it too.

I delete the #-lines and the "override-line".

Until now it works pretty good

Thanks
Peter

pierre83, 2021-01-22:
Hi !
I encountered the same issue (ESP8266 NodeMCU 1.0) with software serial. The issue has been solved by removing the word "override".
So, i removed the #... totally, leaving only "int availableForWrite() {"

Thanks
Pierre

dok-net, 2021-01-25:
Before posting issues, do the minimum research and read the README. I am sorry, but this is an annoying question asked on each and every new release by someone and it's been answered. ESP8266 Arrduino core has a matching EspSoftwareSerial release for each development commit and as such surely for every release. Don't try to mix and match unless you know what you are doing.

closed 2021-01-25 by dok-net