plerup/espsoftwareserial #197: How to use espsoftwareserial with other platforms

ElbyMann, 2021-01-30:
I'm running into a name conflict.  Because the .h file is named "SoftwareSerial.h", my already installed SoftwareSerial (which I use for AVR, etc) is conflicting.  I'm renaming the files in espsoftwareserial, (to EspSoftwareSerial.h and EspSoftwareSerial.h). but doing this manually, and getting out of sync with the repository makes me nervous.  Am I missing a better way of doing this?

ElbyMann, 2021-01-30:
And that doesn't even work.  I'm getting an error message:

/Arduino/code/libraries/EspSoftwareSerial/src/EspSoftwareSerial.h:132:9: error: 'int SoftwareSerial::availableForWrite()' marked override, but does not override

SSBun, 2021-01-31:
@ElbyMann After I removed this coed, it compiled successfully
[image]

ElbyMann, 2021-02-04:
Thanks for your input, SSBun.  It does compile with that change.  Unfortunately, it doesn't work, at least not for me.  Since EspSoftwareSerial seems to be the only non-hardware option for the ESP8266, it looks like I'm going to have to limit my project to the single onboard UART :(

svdrummer, 2021-02-06:
Pretty sure ESPSoftware serial lives in the arduino/15/hardware section.
Google a bit more.

dok-net, 2021-02-06:
@svdrummer @ElbyMann Please read the README coming with this project for information on how to match EspSoftwareSerial releases to ESP8266 releases or developments branches.

closed 2021-02-06 by dok-net