plerup/espsoftwareserial #198: Conflict with definition of atomic variables - TaskManagerIO

rhscdn, 2021-02-09:
I'm seeing a strange definition clash using EspSoftwareSerial and TaskManagerIO in the arduino environment. Hardware is a Wemos D1 Mini (R2) ESP8266 and it occurs whenever you include both TaskManagerIO.h and SoftwareSerial.h. The issue is with each library finding differing definitions for atomic variables that are within scope.

```
/home/user/.platformio/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/include/stdatomic.h:89:0: warning: "ATOMIC_VAR_INIT" redefined [enabled by default]
 #define ATOMIC_VAR_INIT(value)  { .__val = (value) }
 ^
In file included from /home/user/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/atomic:41:0,
...
/home/user/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/bits/atomic_base.h:234:0: note: this is the location of the previous definition
 #define ATOMIC_VAR_INIT(_VI) { _VI }
```

**If I include TaskManagerIO.h before SoftwareSerial.h I get:**

```
Processing d1_mini (platform: espressif8266; board: d1_mini; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 (2.6.2) > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - tool-mklittlefs 1.203.210203 (2.3) 
 - tool-mkspiffs 1.200.0 (2.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TaskManagerIO> 1.1.1
|-- <EspSoftwareSerial> 6.8.5
Building in release mode
Compiling .pio/build/d1_mini/src/activeTest2.cpp.o
In file included from .pio/libdeps/d1_mini/TaskManagerIO/src/TaskPlatformDeps.h:86:0,
                 from .pio/libdeps/d1_mini/TaskManagerIO/src/TaskManagerIO.h:9,
                 from src/activeTest2.cpp:3:
/home/user/.platformio/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/include/stdatomic.h:89:0: warning: "ATOMIC_VAR_INIT" redefined [enabled by default]
 #define ATOMIC_VAR_INIT(value)  { .__val = (value) }
 ^
In file included from /home/user/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/atomic:41:0,
                 from /home/user/.platformio/packages/framework-arduinoespressif8266/libraries/SoftwareSerial/src/circular_queue/circular_queue.h:28,
                 from /home/user/.platformio/packages/framework-arduinoespressif8266/libraries/SoftwareSerial/src/SoftwareSerial.h:27,
                 from src/activeTest2.cpp:2:
/home/user/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/bits/atomic_base.h:234:0: note: this is the location of the previous definition
 #define ATOMIC_VAR_INIT(_VI) { _VI }
 ^
In file included from .pio/libdeps/d1_mini/TaskManagerIO/src/TaskPlatformDeps.h:86:0,
                 from .pio/libdeps/d1_mini/TaskManagerIO/src/TaskManagerIO.h:9,
                 from src/activeTest2.cpp:3:
/home/user/.platformio/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/include/stdatomic.h:380:0: warning: "ATOMIC_FLAG_INIT" redefined [enabled by default]
 #define ATOMIC_FLAG_INIT  { ATOMIC_VAR_INIT(0) }
 ^
In file included from /home/user/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/atomic:41:0,
                 from /home/user/.platformio/packages/framework-arduinoespressif8266/libraries/SoftwareSerial/src/circular_queue/circular_queue.h:28,
                 from /home/user/.platformio/packages/framework-arduinoespressif8266/libraries/SoftwareSerial/src/SoftwareSerial.h:27,
                 from src/activeTest2.cpp:2:
/home/user/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/bits/atomic_base.h:268:0: note: this is the location of the previous definition
 #define ATOMIC_FLAG_INIT { 0 }
 ^
Retrieving maximum program size .pio/build/d1_mini/firmware.elf
Checking size .pio/build/d1_mini/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  34.3% (used 28076 bytes from 81920 bytes)
Flash: [===       ]  26.1% (used 272164 bytes from 1044464 bytes)
```

[... 49 lines cut ...]
/home/user/.platformio/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/include/stdatomic.h:90:36: error: expected ')' before 'void'
/home/user/.platformio/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/include/stdatomic.h:284:18: error: variable or field '__atomic_store_n' declared void
  __atomic_store_n(&(object)->__val, desired, order)
                  ^
/home/user/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/atomic:822:5: error: expected primary-expression before '*' token
     atomic_store_explicit(atomic<_ITp>* __a, _ITp __i,
     ^
/home/user/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/atomic:822:5: error: '__a' was not declared in this scope
     atomic_store_explicit(atomic<_ITp>* __a, _ITp __i,
     ^
/home/user/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/atomic:822:5: error: expected primary-expression before '__i'
     atomic_store_explicit(atomic<_ITp>* __a, _ITp __i,
     ^
/home/user/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/atomic:822:5: error: expected primary-expression before '__m'
     atomic_store_explicit(atomic<_ITp>* __a, _ITp __i,
     ^
/home/user/.platformio/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/include/stdatomic.h:284:18: error: variable or field '__atomic_store_n' declared void
  __atomic_store_n(&(object)->__val, desired, order)
                  ^
/home/user/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/atomic:828:5: error: expected primary-expression before 'volatile'
     atomic_store_explicit(volatile atomic<_ITp>* __a, _ITp __i,
     ^
/home/user/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/atomic:828:5: error: expected ')' before 'volatile'
*** [.pio/build/d1_mini/src/activeTest2.cpp.o] Error 1
```

dok-net, 2021-02-12:
Release 2.6.2, from 28 Nov 2019? TaskManager? Did that even build and run as recently as 2.6.2? Can you prove that this issue has anything to do with EspSoftwareSerial to begin with? Anyway, for supported versions and packages, please use the latest release of ESP8266 Core for Arduino, or the master head, and check the README here for how to deal with matching up EspSoftwareSerial releases to ESP8266 Core for Arduino versions.
Also, for a cooperative, stackful coroutine package, please try [CoopTask](https://www.arduinolibraries.info/libraries/coop-task), I designed that after not finding anything that worked on current ESP8266, and made it cross-platform to ESP32, Windows, Linux, Arduino AVR...

dok-net, 2021-02-21:
@rhscdn No news? More of a janitorial "job" here than fielding real technical support requests, sigh ;-)

rhscdn, 2021-02-22:
Can be closed. Issue resolved by maintainer of TaskManagerIO, Will look at CoopTask.

closed 2021-02-22 by rhscdn