plerup/espsoftwareserial #200: SoftwareSerial::availableForWrite()' marked override, but does not override

rob040, 2021-02-18:
I have replaced esp8266 v2.7.4 included SoftwareSerial library v6.8.5 with latest available here (v6.11.4), but a change caused this compilation error:

```
...\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\SoftwareSerial\src/SoftwareSerial.h:132:9: error: 'int SoftwareSerial::availableForWrite()' marked override, but does not override
     int availableForWrite() override {
         ^
```

I compiled this within the Arduino 1.8.13 IDE on Windows10.

**background:**
I replaced this component in the hope to get better bit timing. I use it to talk to ModBus RTU slave at 9600 half-duplex, that ignores my messages possibly due to framing errors. That slave does accept same messages when send though hardware Uart (using Pro Micro). Scope shows that character timing is off by more than a half bit after 8 bytes and appears to be dependent on bit values; example code Ascii chars bittiming looks fine. I already eliminated I2C code and Wifi code influence.
Could there be an interrupt priority issue?
Are there known issues with 8-bit data with bit7=1?
Any other hints?

dok-net, 2021-02-21:
For matching EspSoftwareSerial to ESP8266 Core for Arduino, this is a repetitive question ad nauseum. RTFM.
I like it that you've taken the timings to a scope, something I've been planning to do for a year or something myself and never got around to. There are some changes in the write timing code between the versions you are referring to. Correlating that to a set bit 7 or a systematic drift would surprise me, but it's not in the realm of impossible. Please install the ESP8266 core from Github master etc and repeat your tests, that way I will be much more interested in the results. In short, any (semi-)professional help is appreciated.

maxgerhardt, 2021-02-26:
This question also popped up on the PlatformIO forums -- I've written [a post here](https://community.platformio.org/t/softwareserial-not-compiling/19578) that shows how to use the library properly; That is, for most users, do not install this library via the library manager, so that the library version used that's contained within the released Arduino-ESP8266 library. It also shows how to use the latest version of the library with the latest Arduino core and compiler, if one really wants. 

So you can also direkt people asking it in the context to PlatformIO to this post without having to explaining too much again.

dok-net, 2021-03-12:
Well, it's, like, open source, so you're free to do pack ports, just don't confuse everyone else with issues and pull requests that are totally besides the point. Please the fine manual, don't mix and match ESP8266 Core for Arduino at your own free will, bad things can happen and you will not get any support from the developers for this, just irony  ;-)

closed 2021-03-12 by dok-net