plerup/espsoftwareserial #204: peekParityError is missing

Pablo2048, 2021-03-23:
Hi,
thanks for the library. It seems like there is discrepancy between readme and real code. In the readme here https://github.com/plerup/espsoftwareserial/issues/new is mentioned `peekParityError()` function, but I can't find it in the code. I think that it can be replaced by https://github.com/plerup/espsoftwareserial/blob/7fdef9be711f703a3e0bd520fc434ce112900f9f/src/SoftwareSerial.h#L142 , but it has to be called AFTER `read()` not before as mentioned in the readme. Am I right, or did I misunderstand something?

dok-net, 2021-03-23:
@Pablo2048 Thanks for catching this. I'll review the whole issue and probably first post a public PR to myself for testing/comments. It will most likely be a breaking change, after all. Again, thank you!

dok-net, 2021-03-23:
@Pablo2048 Please review the PR.

commit 45437b05 (2021-03-26): Fix outdated README, source comments for readParity. Fix dropped 0xff octets. (#205)

* Fixes #203 

* Check for hidden stop bit at the end of each rxBits. Fixes first-octet in session dropped issue, and

intermittently dropped octets, in particular the delay(2000), issue. 

* Fixes #204 

* Use precise enum value identifier in sample code.
closed 2021-03-26 by dok-net