plerup/espsoftwareserial #207: How does transmit work?

dduehren, 2021-04-10:
Your readme on receive is both through and useful.  What I'm wondering is how does sending work.  Is there a corresponding interrupt driven transmit buffer?  Or are my writes blocked until the last byte is sent?  

Thanks

David

dok-net, 2021-04-10:
Write is blocking. I always advertise [CoopTask](https://www.arduino.cc/reference/en/libraries/cooptask/) to anyone seeking to recover processor time from delays or in need of some form of concurrency.

dduehren, 2021-04-11:
How hard would it be to have a transmit buffer that was emptied by interrupt?  I have a half-duplex app where I send several bytes in one direction and get back two or three bytes back.  But I don't like a 4-5 ms hold up while the few bytes I'm sending are semt.

dok-net, 2021-04-11:
If you have a suggestion which interrupt is precise enough and available on the ESP8266 to replace the linear bit-banging code (now at 9600bps 8N1 software serial, writing a single byte invariably blocks for more than one ms) , please show it to me. Otherwise, assuming you are using buffer writes, why not send your bytes one-by-one in a loop and do the other timing-sensitive work between them? Which is exactly what CoopTask would afford you in cooperative multithreading style.

dduehren, 2021-04-11:
Dar,

Thanks for responding.  I have to look up CoopTask to see how I could use it.
Right now I send 3 bytes as 3 single byte writes, one after the other.  I could put this in a state machine. Does the single byte write function return when the byte is written or when the byte has been sent?

Thanks

David Duehren

dduehren, 2021-04-11:
And this type of stuff would be helpful in the readme.  If I send a byte, when does the function return?  The function has to return before the byte is sent otherwise how would a yield get executed? 
And if I want to do my multitasking with state machines and update functions that run in the main loop, doesn’t that work just as well?

Thanks

David Duehren

dok-net, 2021-04-11:
It's software serial, bitbanging can be assumed... please submit any improvements to code and documentation for review. I'm closing this, because explaining how to deal with coroutines or FSM on Arduino is way beyond the scope of this issue tracking.

closed 2021-04-11 by dok-net