plerup/espsoftwareserial #208: ESP32 crash during SPIFFS ops

Csory, 2021-04-15:
**Symptom:** ESP32 (w/Arduino toolchain) crashes if and when circular_queue::push occurs during a SPIFFS read:
"Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
PC: 0x400f0e24: circular_queue ::push(unsigned int&&) at C:\Users\..."

**Cause:** section attribute of a function template is not honored, a [known gcc bug](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=70435). This results in circular_queue:: methods IRAM_ATTR attributes being ignored, and those methods (including ::push) being loaded outside of IRAM (see PC: 0x400f0e24 is not in IRAM indeed...)

**Possible fix:** May I ask the devs to consider modifying circular_queue.h so that the templates would be used as wrappers only, moving the methods outside of them? This would make EspSoftwareSerial functional again on the ESP32 until gcc gets fixed...

dok-net, 2021-04-15:
@Csory Thank you for providing this insight into a technical reason for this known issue, at last there is way to start looking into this seriously.

Csory, 2021-04-15:
Thanks for looking into it! Much older versions - EspSoftwareSerial 5.0.4 and before (the ones without separate circular_queue.h) - are not affected by this gcc bug and work as intended, apparently because of the non-templated code.

dok-net, 2021-04-15:
@csory Please test the PR - simply replacing the updated `circular_queue.h` in your source tree will suffice.

Csory, 2021-04-15:
Just tested it -- works fine now, thanks a lot!

This gcc bug affects lots of 3rd party ESP32 libraries indeed, should be more widely known...

closed 2021-04-15 by Csory

dok-net, 2021-04-18:
Here's code for checking IRAM_ATTR compliance on the ESP8266. Technically, it still fails for this fix, but because the `push()` function is now required to be inlined by the compiler, actually calling, not taking a member function pointer, from a function that is itself in IRAM results in safe code generation by the compiler.
```
#include <SoftwareSerial.h>
#include <circular_queue/circular_queue.h>
void setup()
{
    Serial.begin(115200);
    while (!Serial);
    delay(100);
    Serial.println();
}

void loop()
{
    circular_queue<unsigned char> cq;
    bool (circular_queue<unsigned char>:: * pushPtr)() = &circular_queue<unsigned char>::push;
    // #5780
    // https://github.com/esp8266/esp8266-wiki/wiki/Memory-Map
    if (reinterpret_cast<uint32_t>(reinterpret_cast<void*>(pushPtr)) >= 0x40200000)
    {
        Serial.println(F("function pointer is not in IRAM!"));
    }
    else
    {
        Serial.println(F("function pointer is in IRAM."));
    }
    delay(1000);
}
```

PR #209 (2021-04-15): Inlining push() functions that must be in IRAM for calling from ISRs.
The small overhead is outweighed by
- gcc has a known issue that prevents it from honoring section attributes for function templates
- a possible performance improvement 
Fixes #208.
commit e68ba39c (2021-04-15): Fixes #208. Inlining push() functions that must be in IRAM for calling from ISRs. (#209)