plerup/espsoftwareserial #211: Lib throws error in PlatformIO when building

DreamTexX, 2021-05-27:
Hi, installed this lib in PlatformIO into a project and get the following error:

```
In file included from src/main.cpp:2:0:
.pio/libdeps/env1/EspSoftwareSerial/src/SoftwareSerial.h:132:9: error: 'int SoftwareSerial::availableForWrite()' marked override, but does not override
     int availableForWrite() override {
         ^
.pio/libdeps/env1/EspSoftwareSerial/src/SoftwareSerial.h:159:9: error: 'int SoftwareSerial::read(uint8_t*, size_t)' marked override, but does not override
     int read(uint8_t* buffer, size_t size)
         ^
In file included from .pio/libdeps/env1/EspSoftwareSerial/src/SoftwareSerial.cpp:23:0:
.pio/libdeps/env1/EspSoftwareSerial/src/SoftwareSerial.h:132:9: error: 'int SoftwareSerial::availableForWrite()' marked override, but does not override
     int availableForWrite() override {
         ^
.pio/libdeps/env1/EspSoftwareSerial/src/SoftwareSerial.h:159:9: error: 'int SoftwareSerial::read(uint8_t*, size_t)' marked override, but does not override
     int read(uint8_t* buffer, size_t size)
         ^
```

How can I fix it?

dok-net, 2021-05-27:
ESP8266 Core For Arduino ships with an included EspSoftwareSerial library, such that APIs match. In Arduino IDE, you must not install an EspSoftwareSerial library via the library manager at all.
I don't know about Platformio, what steps did you undertake to "install this lib"? Without  having been given rationale, method, and version number you used, my best reply would be to ask for all of that, and if you know more about Platformio library management, in particular, setting up version dependencies to BSPs, please look into that and find out how to set up EspSoftwareSerial's descriptor in such a way that for future releases the Platformio dependency resolver will be able to pick only compatible versions.

For the lazy solution: If you don't need the latest release of EspSoftwareSerial, 6.11.2 would be the latest that compiles with the Platformio ESP8266 release out of the box; otherwise, this is the respective patch you could revert:
```
diff --git a/src/SoftwareSerial.h b/src/SoftwareSerial.h
index ea93f2f..15a4e79 100644
--- a/src/SoftwareSerial.h
+++ b/src/SoftwareSerial.h
@@ -156,9 +156,13 @@ public:
         return (0x9669 >> byte) & 1;
     }
     /// The read(buffer, size) functions are non-blocking, the same as readBytes but without timeout
-    size_t read(uint8_t* buffer, size_t size);
+    int read(uint8_t* buffer, size_t size)
+//#if defined(ESP8266)
+//        override
+//#endif
+        ;
     /// The read(buffer, size) functions are non-blocking, the same as readBytes but without timeout
-    size_t read(char* buffer, size_t size) {
+    int read(char* buffer, size_t size) {
         return read(reinterpret_cast<uint8_t*>(buffer), size);
     }
     /// @returns The number of bytes read into buffer, up to size. Times out if the limit set through
```
```
diff --git a/src/SoftwareSerial.cpp b/src/SoftwareSerial.cpp
index 32a162b..0208e10 100644
--- a/src/SoftwareSerial.cpp
+++ b/src/SoftwareSerial.cpp
@@ -204,9 +204,9 @@ int SoftwareSerial::read() {
     return val;
 }
 
-size_t SoftwareSerial::read(uint8_t* buffer, size_t size) {
+int SoftwareSerial::read(uint8_t* buffer, size_t size) {
     if (!m_rxValid) { return 0; }
-    size_t avail;
+    int avail;
     if (0 == (avail = m_buffer->pop_n(buffer, size))) {
         rxBits();
         avail = m_buffer->pop_n(buffer, size);
```

DreamTexX, 2021-05-27:
I've just added `plerup/espsoftwareserial @ ^6.12.4` to the lib_deps in the platformio.ini file, then vs code updated the dependencies and downloaded the version. (Using 6.12.5 does not work, as it downloads nothing).

I do not know much more about platformio or platformio dependency management as I am relatively new to platformio.

Also, I just discovered, platformio is already using this library with version 6.8.5 (in the `framework-arduinoespressif8266` package), but replacing this with the newest version (1.12.4 or even 1.12.5) results in the same error.

dok-net, 2021-05-27:
@DreamTexX Please pay more attention to the version numbers, re-read you last message :-)
Anyway, if you can, follow my advice to modify the source code, if you don't know how, I don't know how I can help you, beyond asking you, why you started mix and matching (non-matching actually) library and platform versions in the first place.

DreamTexX, 2021-05-28:
> 1.12.4 or even 1.12.5

Haha, of course, I meant 6.12.4 and 6.12.5.

> why you started to mix and matching (non-matching actually) library and platform versions

I don't really see where I am mixing library and platform versions, I just install espsoftwareserial 6.12.4 over espsoftwareserial 6.8.5. That causes these errors. Then I, instead of changing the library version in the native platformio package for the esp8266, I installed 6.12.4 as a direct dependency of the project. Same error.

DreamTexX, 2021-05-28:
> 6.11.2 would be the latest that compiles with the Platformio ESP8266 release out of the box

This, btw, does not work too when adding this version as dep to the project. I really have no clue what I am doing wrong.

dok-net, 2021-05-28:
Hum, Platformio calls Arduino a "framework". I would like to think that is stays as close as possible to the way Arduino works, such that there are "cores" supporting "boards". That said, the ESP8266 Core for Arduino always includes a SoftwareSerial library in its "libraries" subdirectory, in Git, it is a submodule. This expresses, and provides, a fixed dependency. If you check the documentation, it says that you are not supposed to install externally the EspSoftwareSerial library. Honestly, I don't know what Platformio is doing, but it shouldn't stray too far from that, because that is how the development of the core and EspSoftwareSerial is taking place. That's what I meant by mix-and-match, you just shall not touch this stuff at all - if you want to update to a later EspSoftwareSerial library, provide a compelling reason, and if you want to stay within supported boundaries, use Github to update the ESP8266 core and let that pull in an EspSoftwareSerial version that matches. Anything else, you are on your own (seriously).

I am closing this issue, because as you see, it's a non-issue, besides an expectation that is clearly rejected by existing documentation. If Platformio gives you a different (wrong) impression, please report there. You see, on each new release I am fielding this type of question here, really no offense, but it is tiresome.

By all means, whenever you find issues with how the library works, report that here. I am not saying, that you can never ask about compilation problems, too, but manually switching versions is just not a supported activity.

closed 2021-05-28 by dok-net