plerup/espsoftwareserial #212: Lib gives CRC error when trying to upgrade in IDE library manager

RobTillaart, 2021-05-27:
See screenshot library manager
IDE 1.8.13

[image]

dok-net, 2021-05-27:
> CRC doesn't match, file is corrupted. It may be a temporary problem, please retry later.

Did you follow the instructions? Anyway, this is not an issue with the library proper, but rather the Arduino library archives etc.

dok-net, 2021-05-27:
Oh, and never use the Arduino library manager to install EspSoftwareSerial for ESP8266, if that's your platform, because it's included in ESP8266 Core For Arduino and you please must read the instructions, it describes what and what not you can do.

closed 2021-05-27 by dok-net