plerup/espsoftwareserial #213: SoftwareSerial.h not working with ESP32 S2 have any solution?

amin-is, 2021-06-03:


svdrummer, 2021-06-03:
Can you provide something to work off? All you have said is YOU have a problem 😀 Any errors? Does it compile, code snippet...anything.

dok-net, 2021-06-03:
While this "issue" report doesn't meet any quality standards and I should reject it immediately, I got intrigued and, for lack of hardware, compiled EspSoftwareSerial Git master head, [ESP32 Arduino Core](Arduino https://github.com/espressif/arduino-esp32.git) master head, using the EspSoftwareSerial`s `swsertest` example sketch, for the "ESP32S2 Dev Module" board selection.
I can report that it builds without errors.
If the original reporter could provide their findings based on that preselection, I would be willing to look into that, otherwise I will close this as invalid in due time.

amin-is, 2021-06-03:
> While this "issue" report doesn't meet any quality standards and I should reject it immediately, I got intrigued and, for lack of hardware, compiled EspSoftwareSerial Git master head, [ESP32 Arduino Core](https://github.com/espressif/arduino-esp32.git) master head, using the EspSoftwareSerial's `swsertest` example sketch, for the "ESP32S2 Dev Module" board selection.
> I can report that it builds without errors.
> If the original reporter could provide their findings based on that preselection, I would be willing to look into that, otherwise I will close this as invalid in due time.

Please don't close i have added my error log above.

dok-net, 2021-06-03:
Dear @amin-is , it's OK to be a newcomer to this, I hope you'll take away some lessons from this.
All I can say is that as far as I am concerned the support for ESP32S2 has been landed in the abovementioned Github repository, whether there is a release to consume via the Arduino IDE boards manager or not is beyond my interest in these things (ESP32 core developement team isn't much interested in EspSoftwareSerial due to the multiple HW UARTS in their SoC - does that hold for the S2, I don't know). If you follow my lead - and please consult the various other only resources on getting current development bits for the ESP32 from Github working on your workstation - you will be able to compile.
I that regard, no offense, this issue is determined invalid and I'll close it.
Please do come back with a proper issue, only if and when you have applied due diligence. You can research so many previous issue reports by those before you, even here, just remove the `is:open` filter setting.
By the way, you may find some advise on how to use the HW UARTs is this way, as well.

closed 2021-06-03 by dok-net

tchilton, 2021-07-07:
Please see #219  for info on this problem and the solution to make it work.