plerup/espsoftwareserial #216: Corrupted/truncated data on ESP32

I-Connect, 2021-06-09:
Hi,

I am trying to read serial data on a ESP32 from a Victron power converter, this puts out data blocks once a second of about 131 bytes at 19200 baud. (it works with HW serial for test but I am using all serials for other devices)

I reverted to a simple test script for trouble shooting.

```
#include "Arduino.h"
#include "SoftwareSerial.h"

SoftwareSerial veDirectSwSerial;

void setup() {
    
  Serial.begin(115200);
  veDirectSwSerial.begin(19200,SWSERIAL_8N1, GPIO_NUM_15, GPIO_NUM_12, false, 256, 28 );
}

void loop() {
    delay(5000);

    while (veDirectSwSerial.available()) {
      log_d("%c", veDirectSwSerial.read());
    }
}

```
Output I get varies:
-You see multiple times that only the first block is read (starting with "P I") while the buffer and delay should be large enough to capture a complete block of data in the buffer.
-Corrupted chars: after "P I" a "D" should follow instead of �
-And you see below sometimes I get a larger block of data which is almost oke but still has some corrupted chars. After that I again only get the first few bytes.

Any hints what is going/I am doing wrong? thx!

```
[D][main.cpp:28] loop(): P
[D][main.cpp:28] loop(): I
[D][main.cpp:28] loop(): �
[D][main.cpp:28] loop(): 
[D][main.cpp:28] loop():

[D][main.cpp:28] loop(): P
[D][main.cpp:28] loop(): I
[D][main.cpp:28] loop(): �
[D][main.cpp:28] loop():
[D][main.cpp:28] loop():

[D][main.cpp:28] loop(): P
[D][main.cpp:28] loop(): I
[D][main.cpp:28] loop(): D
[D][main.cpp:28] loop():
[D][main.cpp:28] loop(): 0
[D][main.cpp:28] loop(): x
[D][main.cpp:28] loop(): A
[D][main.cpp:28] loop(): 2
[D][main.cpp:28] loop(): 5
[D][main.cpp:28] loop(): 
[D][main.cpp:28] loop():
[D][main.cpp:28] loop(): 

[D][main.cpp:28] loop(): F
[D][main.cpp:28] loop(): W
[D][main.cpp:28] loop():
[D][main.cpp:28] loop(): 0
[D][main.cpp:28] loop(): 1
[D][main.cpp:28] loop(): 2
[D][main.cpp:28] loop(): 1
[D][main.cpp:28] loop():
[D][main.cpp:28] loop():

[D][main.cpp:28] loop(): S
[D][main.cpp:28] loop(): H
[D][main.cpp:28] loop(): Q
[D][main.cpp:28] loop(): 2
[D][main.cpp:28] loop(): 0
[D][main.cpp:28] loop(): 1
[D][main.cpp:28] loop(): 6
[D][main.cpp:28] loop(): Z
[D][main.cpp:28] loop(): D
[D][main.cpp:28] loop(): 1
[D][main.cpp:28] loop(): H
[D][main.cpp:28] loop():
[D][main.cpp:28] loop():

[D][main.cpp:28] loop(): M
[D][main.cpp:28] loop(): O
[D][main.cpp:28] loop(): 2
[D][main.cpp:28] loop():
[D][main.cpp:28] loop(): 

[D][main.cpp:28] loop(): C
[D][main.cpp:28] loop(): S
[D][main.cpp:28] loop():
[D][main.cpp:28] loop(): 9
[D][main.cpp:28] loop():
[D][main.cpp:28] loop(): 

[D][main.cpp:28] loop(): A
[D][main.cpp:28] loop(): C
[D][main.cpp:28] loop(): _
[D][main.cpp:28] loop(): O
[D][main.cpp:28] loop(): U
[D][main.cpp:28] loop(): Y
[D][main.cpp:28] loop(): 2
[D][main.cpp:28] loop(): 3
[D][main.cpp:28] loop(): 0
[D][main.cpp:28] loop(): 0
[D][main.cpp:28] loop(): 7
[D][main.cpp:28] loop():
[D][main.cpp:28] loop(): 

[D][main.cpp:28] loop(): A
[D][main.cpp:28] loop(): C
[D][main.cpp:28] loop(): _
[D][main.cpp:28] loop(): O
[D][main.cpp:28] loop(): U
[D][main.cpp:28] loop(): T
[D][main.cpp:28] loop(): 0
[D][main.cpp:28] loop(): V
[D][main.cpp:28] loop():
[D][main.cpp:28] loop(): 1
[D][main.cpp:28] loop(): 2
[D][main.cpp:28] loop(): 3
[D][main.cpp:28] loop(): 0
[D][main.cpp:28] loop(): 7
[D][main.cpp:28] loop():
[D][main.cpp:28] loop():

[D][main.cpp:28] loop(): A
[D][main.cpp:28] loop(): R
[D][main.cpp:28] loop():
[D][main.cpp:28] loop(): �
[D][main.cpp:28] loop(): I
[D][main.cpp:28] loop(): 9
[D][main.cpp:28] loop(): 0
[D][main.cpp:28] loop():
[D][main.cpp:28] loop():

[D][main.cpp:28] loop(): O
[D][main.cpp:28] loop(): R
[D][main.cpp:28] loop():
[D][main.cpp:28] loop(): 0
[D][main.cpp:28] loop(): x
[D][main.cpp:28] loop(): �
[D][main.cpp:28] loop(): 0
[D][main.cpp:28] loop(): 0
[D][main.cpp:28] loop(): 0
[D][main.cpp:28] loop(): 0
[D][main.cpp:28] loop(): 0
[D][main.cpp:28] loop(): 0
[D][main.cpp:28] loop(): 0
[D][main.cpp:28] loop():
[D][main.cpp:28] loop(): 

[D][main.cpp:28] loop(): C
[D][main.cpp:28] loop(): h
[D][main.cpp:28] loop(): e
[D][main.cpp:28] loop(): �
[D][main.cpp:28] loop(): �
[D][main.cpp:28] loop(): ␙
[D][main.cpp:28] loop(): 
[D][main.cpp:28] loop():

[D][main.cpp:28] loop(): P
[D][main.cpp:28] loop(): I
[D][main.cpp:28] loop(): �
[D][main.cpp:28] loop(): 
[D][main.cpp:28] loop():

[D][main.cpp:28] loop(): P
[D][main.cpp:28] loop(): I
[D][main.cpp:28] loop(): �
[D][main.cpp:28] loop(): 
[D][main.cpp:28] loop():

[D][main.cpp:28] loop(): P
[D][main.cpp:28] loop(): I
[D][main.cpp:28] loop(): �
[D][main.cpp:28] loop(): 
[D][main.cpp:28] loop():

[D][main.cpp:28] loop(): P
[D][main.cpp:28] loop(): I
[D][main.cpp:28] loop(): �
[D][main.cpp:28] loop(): 
[D][main.cpp:28] loop():
```

dok-net, 2021-06-09:
It's all about a complete misunderstanding what the last parameter to `begin()` affects. Please remove it and report your results.

I-Connect, 2021-06-10:
Hi, thx for the quick response.

I removed the last parameter and now data is passed/read as expected.

To be honest I did read the docs but had a hard time understanding what that last parameter exactly meant and what value to use for it.
But I completely overlooked this sentence in the example: "Do not copy this into your own project without reading the documentation." :-( sorry for that, it should have triggered me.

I added the software serial reading the victron to my complete project where the ESP has tons to do (CAN, multiple I2C devices, GSM via serial, Nextion display via serial, BLE scanning and several other tasks running logic) and at first glance it seems to works flawlessly. :-)

Great library, thx!

closed 2021-06-10 by dok-net