plerup/espsoftwareserial #218: May Please update espsoftwareserial for esp32s2 esp32c3

cephaswang, 2021-07-07:
May Please update espsoftwareserial for esp32s2 esp32c3

yanbwang, 2021-07-07:
I have tried to compile example in "espsoftwareserial-master\examples\loopback" with ESP32 S2. And got below error message:

"C:\Users\DaveD\OneDrive - Intel Corporation\Documents\s2esp32\hardware\espressif\esp32/tools/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-ar" cru  "C:\Users\DaveD\AppData\Local\Temp\arduino_build_701025\core\core.a" "C:\Users\DaveD\AppData\Local\Temp\arduino_build_701025\core\cbuf.cpp.o"
"C:\Users\DaveD\OneDrive - Intel Corporation\Documents\s2esp32\hardware\espressif\esp32/tools/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-ar" cru  "C:\Users\DaveD\AppData\Local\Temp\arduino_build_701025\core\core.a" "C:\Users\DaveD\AppData\Local\Temp\arduino_build_701025\core\main.cpp.o"
Archiving built core (caching) in: C:\Users\DaveD\AppData\Local\Temp\arduino_cache_997350\core\core_espressif_esp32_esp32s2_SerialMode_default,UploadMode_default,PSRAM_enabled,PartitionScheme_default,CPUFreq_240,FlashMode_qio,FlashFreq_80,FlashSize_4M,UploadSpeed_921600,DebugLevel_none_bfbed8c645d4fd65928d02b2acf492f2.a
Linking everything together...
"C:\Users\DaveD\OneDrive - Intel Corporation\Documents\s2esp32\hardware\espressif\esp32/tools/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++" "-Wl,--Map=C:\Users\DaveD\AppData\Local\Temp\arduino_build_701025/loopback.ino.map" "-LC:\Users\DaveD\OneDrive - Intel Corporation\Documents\s2esp32\hardware\espressif\esp32/tools/sdk/esp32s2/lib" "-LC:\Users\DaveD\OneDrive - Intel Corporation\Documents\s2esp32\hardware\espressif\esp32/tools/sdk/esp32s2/ld" -mlongcalls -T esp32s2.rom.ld -T esp32s2.rom.libgcc.ld -T esp32s2.rom.newlib-data.ld -T esp32s2.rom.newlib-funcs.ld -T esp32s2.rom.spiflash.ld -T esp32s2_out.ld -T esp32s2.project.ld -T esp32s2.peripherals.ld  -Wl,--cref -fno-rtti -fno-lto -u esp_app_desc -u vfs_include_syscalls_impl -u pthread_include_pthread_impl -u pthread_include_pthread_cond_impl -u pthread_include_pthread_local_storage_impl -u app_main -u call_user_start_cpu0 -u ld_include_panic_highint_hdl -Wl,--gc-sections -Wl,--undefined=uxTopUsedPriority -u newlib_include_locks_impl -u newlib_include_heap_impl -u newlib_include_syscalls_impl -u newlib_include_pthread_impl -u __cxa_guard_dummy  -Wl,--start-group "C:\Users\DaveD\AppData\Local\Temp\arduino_build_701025\sketch\loopback.ino.cpp.o" "C:\Users\DaveD\AppData\Local\Temp\arduino_build_701025\libraries\espsoftwareserial-master\SoftwareSerial.cpp.o" "C:\Users\DaveD\AppData\Local\Temp\arduino_build_701025\core\core.a" -lxtensa -lmbedtls -lefuse -lbootloader_support -lapp_update -lesp_ipc -lspi_flash -lesp_system -lsoc -lvfs -lesp_eth -ltcpip_adapter -lesp_netif -lesp_event -lwpa_supplicant -lnvs_flash -lesp_wifi -llwip -llog -lheap -lesp_ringbuf -ldriver -lpthread -lespcoredump -lesp32s2 -lesp_common -lesp_timer -lfreertos -lnewlib -lcxx -lapp_trace -lasio -lcbor -lcoap -lconsole -lnghttp -lesp-tls -lesp_gdbstub -lesp_hid -ltcp_transport -lesp_http_client -lesp_http_server -lesp_https_ota -lesp_https_server -lprotobuf-c -lprotocomm -lmdns -lesp_local_ctrl -lsdmmc -lesp_serial_slave_link -lesp_websocket_client -lexpat -lwear_levelling -lfatfs -lfreemodbus -ljsmn -ljson -llibsodium -lmqtt -lopenssl -lperfmon -lspiffs -ltinyusb -lulp -lunity -lwifi_provisioning -lfb_gfx -lasio -lcbor -lcoap -lesp_gdbstub -lesp_hid -lesp_https_ota -lesp_local_ctrl -lesp_https_server -lesp_serial_slave_link -lesp_websocket_client -lexpat -lfreemodbus -ljsmn -llibsodium -lmqtt -lperfmon -lunity -lwifi_provisioning -lprotocomm -lprotobuf-c -ljson -lfb_gfx -lmdns -lconsole -lfatfs -lsdmmc -lwear_levelling -lopenssl -lspiffs -ltinyusb -lxtensa -lmbedtls -lefuse -lbootloader_support -lapp_update -lesp_ipc -lspi_flash -lesp_system -lsoc -lvfs -lesp_eth -ltcpip_adapter -lesp_netif -lesp_event -lwpa_supplicant -lnvs_flash -lesp_wifi -llwip -llog -lheap -lesp_ringbuf -ldriver -lpthread -lespcoredump -lesp32s2 -lesp_common -lesp_timer -lfreertos -lnewlib -lcxx -lapp_trace -lnghttp -lesp-tls -ltcp_transport -lesp_http_client -lesp_http_server -lulp -lmbedtls -lmbedcrypto -lmbedx509 -lsoc_esp32s2 -lcoexist -lcore -lespnow -lmesh -lnet80211 -lpp -lrtc -lsmartconfig -lphy -lxtensa -lmbedtls -lefuse -lbootloader_support -lapp_update -lesp_ipc -lspi_flash -lesp_system -lsoc -lvfs -lesp_eth -ltcpip_adapter -lesp_netif -lesp_event -lwpa_supplicant -lnvs_flash -lesp_wifi -llwip -llog -lheap -lesp_ringbuf -ldriver -lpthread -lespcoredump -lesp32s2 -lesp_common -lesp_timer -lfreertos -lnewlib -lcxx -lapp_trace -lnghttp -lesp-tls -ltcp_transport -lesp_http_client -lesp_http_server -lulp -lmbedtls -lmbedcrypto -lmbedx509 -lsoc_esp32s2 -lcoexist -lcore -lespnow -lmesh -lnet80211 -lpp -lrtc -lsmartconfig -lphy -lxtensa -lmbedtls -lefuse -lbootloader_support -lapp_update -lesp_ipc -lspi_flash -lesp_system -lsoc -lvfs -lesp_eth -ltcpip_adapter -lesp_netif -lesp_event -lwpa_supplicant -lnvs_flash -lesp_wifi -llwip -llog -lheap -lesp_ringbuf -ldriver -lpthread -lespcoredump -lesp32s2 -lesp_common -lesp_timer -lfreertos -lnewlib -lcxx -lapp_trace -lnghttp -lesp-tls -ltcp_transport -lesp_http_client -lesp_http_server -lulp -lmbedtls -lmbedcrypto -lmbedx509 -lsoc_esp32s2 -lcoexist -lcore -lespnow -lmesh -lnet80211 -lpp -lrtc -lsmartconfig -lphy -lxtensa -lmbedtls -lefuse -lbootloader_support -lapp_update -lesp_ipc -lspi_flash -lesp_system -lsoc -lvfs -lesp_eth -ltcpip_adapter -lesp_netif -lesp_event -lwpa_supplicant -lnvs_flash -lesp_wifi -llwip -llog -lheap -lesp_ringbuf -ldriver -lpthread -lespcoredump -lesp32s2 -lesp_common -lesp_timer -lfreertos -lnewlib -lcxx  [... cut ...] its()':

c:\users\DaveD\onedrive - intel corporation\documents\s2esp32\hardware\espressif\esp32\tools\xtensa-esp32s2-elf\xtensa-esp32s2-elf\include\c++\8.2.0\bits/atomic_base.h:413: undefined reference to `__atomic_exchange_1'

collect2.exe: error: ld returned 1 exit status
Using library espsoftwareserial-master at version 6.12.7 in folder: C:\Users\DaveD\OneDrive - Intel Corporation\Documents\s2esp32\libraries\espsoftwareserial-master 
exit status 1
Error compiling for board ESP32S2 Dev Module.

dok-net, 2021-07-07:
Arduino Core for ESP32 from Github master, properly set up, and using EspSoftwareSerial release 6.12.7, doing a clean build from scratch, `examples/loopback./loopback.ino` also compiles perfect. For both ESP32S2 Dev Module and ESP32C3 Dev Module.

closed 2021-07-07 by dok-net

tchilton, 2021-07-07:
Hi, I've had similar issues, please see defect #219 where I provide info on the fixes to make this work. I hope that the  project will incorporate my suggestions

yanbwang, 2021-07-08:
> Arduino Core for ESP32 from Github master, properly set up, and using EspSoftwareSerial release 6.12.7, doing a clean build from scratch, `examples/loopback./loopback.ino` also compiles perfect. For both ESP32S2 Dev Module and ESP32C3 Dev Module.

Thanks for your guide. I will try it once I update to lastest ESP32 S2 core.

dok-net, 2021-07-08:
@yanbwang It seems that you own the actual hardware, please participate in fixing the GPIO availability logic in issue report #219.

yanbwang, 2021-07-08:
> @yanbwang It seems that you own the actual hardware, please participate in fixing the GPIO availability logic in issue report #219.

I have tried, but I can't find the code described in #219. I guess it's caused by the ESP32 S2 environment which is too old. I'm working on a USB project which latest ESP32 Arduino has some issues. Thanks.

tchilton, 2021-07-08:
@yanbwang 
The instructions for updating the IDF framework, compiler and tools is in the link at (https://www.mischianti.org/2020/12/01/esp32-s2-pinout-specs-and-arduino-ide-configuration-1/)

I've submitted a pull request with my edits to clean up the support for ESP32-S2 and ESP32-C3.
With both of those in place, I have a working SoftwareSerial on the ESP32-S2. I do not have any ESP32-C3  to test with