plerup/espsoftwareserial #222: GPIO15 removed as usable pin

I-Connect, 2021-08-25:
Hi,

I encountered an issue within my application with espsoftwareserial versions > 6.12.6 . After checking I see you have removed GPIO15 as a usable pin.

I know this is a strapping pin but when used as RX there is no issue. GPIO15 only outputs PWM at boot after that it is fine to use as RX.
Can you please add this pin again? 

You also removed GPIO12, this one is more "dangerous" to use. But still usable if you burn the efuse to ignore this pin at boot.

I guess the developer of the hardware/application should determine if a pin can/should be used or not? There are more then 1 work arounds for the strapping pins and maybe they are needed if you run out of IO... (like I do :-) )

thx!

closed 2021-08-29 by dok-net

dok-net, 2021-08-30:
@tchilton FYI - if so needed, please adapt your code using the library to handle the GPIOs that are now, again, available on the ESP32 after this rollback.

tchilton, 2021-08-31:
Thanks for the notification.

Referring back to our previous discussions on if the safety checks are needed vs knowing the chip and its limitations, I agree with i-Connect about who's responsible for design decisions about each specific implementation.  The GPIOI12 debate being a good case in point.

Perhaps a decision for your library - do you go fully safe to protect more novice users so that they don't fall into unexpected issues vs going fully unsafe for people that understand the chip better and have probably read the data sheet and have a better understanding on whats going on.

So, a suggestion - you could add a "safety catch" type method in the class - by default gpio_safety mode is on and all unsafe pins are disabled, but if you flip the safety catch to turn gpio_safety off, then the less safe pins become available. This would allow you to satisfy both types of users and give some warning that what they are doing is a bit more advanced. You could even include the links such as the one that I-Connect provided so that people know where to read up on the specifics of the pins being enabled and what risks they are taking. The same thing could optionally be done via another parameter in the constructor since you allow config via multiple methods.

The only can of worms that opens up is what level of risk do you then allow ? 

Take the strapping pins as good examples - it would be easy for people to get to non-bootable ESP's if there is serial data flowing to the ESP at bootup / during a brownout / watchdog event etc and it senses the boot pins as indicating the firmware update mode for example, so the device wouldn't boot properly.

dok-net, 2021-09-01:
> The only **can of worms** that opens up is what level of risk do you then allow

I'm not sure and don't much care how much effort the other libraries out there go to to pseudo-protect users from messing up their MCUs from the software side :-) Thank you for your thoughts, someone researching the issue hopefully with find these discussions, but unless there's something technically wrong with the exact GPIOs ruled in or out, I'm not going down this direction any further in this free (yes, as in beer ;-) ) effort.

commit 4b819b09 (2021-08-29): Permit use of GPIO12 on ESP32 and add link to the specifics in README.md

Fixes #222