plerup/espsoftwareserial #227: SoftwareSerial sometimes stalls for a exceedingly large time under ESP32-S2 @9600 baud

avillacis, 2021-11-21:
Hardware: ESP32-S2 (YUBOX One), 2MB PSRAM, Nextion screen on pins 5 (TX), 7 (RX)
Software: Arduino IDE 1.8.16, Arduino-ESP32 2.0.1 (released 9 November 2021), espsoftwareserial 6.15.1

Under certain conditions (possibly if MCU cannot keep up with serial bit timing?), a delay() call will be fed an enormously large positive value, causing the entire task to stall when attempting to write a byte sequence.

This issue was found while testing stability of the Espsoftwareserial library on ESP32-S2 while driving a Nextion screen at 9600 baud. Suddenly, with no previous warning, the SoftwareSerial.write() call can spontaneously stall without returning from the write() call. As deduced from continuing progress by other lower-priority tasks, the task attempting the write is not in a busy loop, but in a delay() call or equivalent. From this, I placed this diagnostic patch on the library:

```diff
diff -ur /home/alex/checkouts/arduino/esp32/plerup/espsoftwareserial-git/src/SoftwareSerial.cpp ./src/SoftwareSerial.cpp
--- /home/alex/checkouts/arduino/esp32/plerup/espsoftwareserial-git/src/SoftwareSerial.cpp	2021-11-15 11:29:32.011520195 -0500
+++ ./src/SoftwareSerial.cpp	2021-11-21 12:34:22.618058479 -0500
@@ -293,6 +293,9 @@
         const int32_t ms = remaining / 1000L / ESP.getCpuFreqMHz();
         if (ms > 0)
         {
+            if (ms > 200) {
+                log_e("ms=%d , why??? m_periodStart=%u expired=%u remaining=%d", ms, m_periodStart, expired, remaining);
+            }
             delay(ms);
         }
         else
```

When the bug triggers, the patch shows this output:

```
[5520055][E][SoftwareSerial.cpp:297] preciseDelay(): ms=17895697 , why??? m_periodStart=1853336517 expired=28065 remaining=-3065

```
Note the large value of 17895697 as the final result of the division, which is then used as the argument for the delay() function, causing the stall.

avillacis, 2021-11-21:
I think that the root cause is that the result of `remaining / 1000L` is signed, but is being cast to unsigned (uint32_t) before being again divided by the result of ESP.getCpuFreqMHz() (which returns an uint32_t), yielding an unsigned result before being assigned to `ms`. If the value of `remaining` is equal or below -1000, the entire expression is somewhat equivalent to `((uint32_t)(-1)) / ESP.getCpuFreqMHz()`.

avillacis, 2021-11-21:
I propose this one-line fix for this bug:

```diff
diff -ur /home/alex/checkouts/arduino/esp32/plerup/espsoftwareserial-git/src/SoftwareSerial.cpp ./src/SoftwareSerial.cpp
--- /home/alex/checkouts/arduino/esp32/plerup/espsoftwareserial-git/src/SoftwareSerial.cpp	2021-11-15 11:29:32.011520195 -0500
+++ ./src/SoftwareSerial.cpp	2021-11-21 16:13:54.214076930 -0500
@@ -290,7 +290,7 @@
         if (!m_intTxEnabled) { xt_wsr_ps(m_savedPS); }
         const auto expired = ESP.getCycleCount() - m_periodStart;
         const int32_t remaining = m_periodDuration - expired;
-        const int32_t ms = remaining / 1000L / ESP.getCpuFreqMHz();
+        const int32_t ms = (remaining > 0) ? remaining / 1000L / ESP.getCpuFreqMHz() : 0;
         if (ms > 0)
         {
             delay(ms);
```

dok-net, 2021-11-21:
@avillacis Thanks for pointing this out. I'll rethink the use of `ESP.getCpuFreqMhz()` in general, whenever in the context of signed divisions. Off the top of my head, I propose this instead:
`const int32_t ms = remaining / 1000L / (int32_t)ESP.getCpuFreqMHz();`

What do you think?

avillacis, 2021-11-22:
> @avillacis Thanks for pointing this out. I'll rethink the use of `ESP.getCpuFreqMhz()` in general, whenever in the context of signed divisions. Off the top of my head, I propose this instead as a common fix:
> 
> ```diff
> -        const int32_t ms = remaining / 1000L / ESP.getCpuFreqMHz();
> +        const int32_t ms = remaining / 1000L / static_cast<int32_t>(ESP.getCpuFreqMHz());
> ```
> 
> What do you think?

I rather prefer the fix I initially proposed. Instead of depending on signed/unsigned division rules, my proposal explicitly checks for a negative value of `remaining` and makes clear that the code should not wait if the wait time has already been exceeded.

BTW, the entire library appears to write data correctly when used in an ESP32 (2-core), but frequently corrupts outgoing data when used under ESP32-S2 (1-core), especially when WiFi is in use. Raising the priority of the task that writes to the SoftwareSerial is not helping with the data corruption. I am still making experiments before opening a new bug. Any ideas?

dok-net, 2021-11-23:
Given that tx is synchronous bit-banging, I can only imagine that the S2 has performance issues - even the ESP32 gives worse maximum performance compared to the ESP8266, I attribute it to the use of the RTOS. You are more than welcome to analyse and provide insights into anything that can be fixed.

dok-net, 2021-11-24:
Have you tried `enableIntTx(false);`?

avillacis, 2021-11-26:
> Have you tried `enableIntTx(false);`?

I have not. From what I can see here:

https://github.com/plerup/espsoftwareserial/blob/cff4deb789ebd05d60a2eff5258693664d52f8a9/src/SoftwareSerial.cpp#L25-L30

The only effect of enableIntTx() is to enable or disable use of the `xt_rsil` and `xt_wsr_ps` macros, and these macros are no-ops when compiling under ESP32.

dok-net, 2021-11-27:
@avillacis Please try https://github.com/plerup/espsoftwareserial/commit/b1aae5c330cb25785071cd78a574c240591c63ed

closed 2021-11-27 by dok-net

dok-net, 2021-11-27:
@avillacis Please test whether eb4b29074b75eacac3585bf84b5495b8f80a92cf fixes the issue you mentioned on ESP32 S2.

avillacis, 2021-11-29:
> @avillacis Please test whether [eb4b290](https://github.com/plerup/espsoftwareserial/commit/eb4b29074b75eacac3585bf84b5495b8f80a92cf) fixes the issue you mentioned on ESP32 S2. On the ESP32 C3, sending bit patterns between Serial1 and EspSoftwareSerial works fine at 115200bps, 8N1, half-duplex.

With 6.15.2, using enableIntTx(false) now fixes the ESP32-S2 TX corruption issue when using my Nextion at 9600 baud while doing other things including having WiFi on. Thanks! I have not tried at any higher speed for now.

commit b1aae5c3 (2021-11-27): Mixed signed/unsigned division bug. Fixes #227