plerup/espsoftwareserial #232: unexpected interrupts occurred

leoujz, 2022-03-22:
since I set the interrupt triggering condition as RISE, sometimes digitalRead(pin) returns 0 in the interrupt handler, though I saw a very clean wave in the oscilloscope. I know that there must be some unexpected voltage fluctuation which is not captured by our eyes, and sometimes the actual communication won't be that ideal. is the algorithm too sensitive or the interrupt is too sensitive?
Is there a software way to reduce the sensitivity?

dok-net, 2022-03-26:
> since I set the interrupt triggering condition as RISE

Is this a general question regarding interrupts? I don't see how triggering only on rise applies to the serial protocol :-)

leoujz, 2022-03-27:
I'm sorry for not describing it in detail. I see that serial protocol contains continuous high and low level, because I did not get right result and try to figure out why, I just set it to RISE instead of CHANGE to check the interrupt is working well. 
And another reason I set it to RISE is that, the signal I was processing use half high half low level to represent 1, instead of all high level as serial protocol, so I will get a RISE for each bit with value 1, and if no RISE that must be a 0 value.

dok-net, 2022-04-09:
I think there's mitigating logic in place; unless you have actual problems with regular EIA-232 data, let's close this.

closed 2022-04-12 by leoujz