plerup/espsoftwareserial #233: Espsoftserial on FreeRTOS task problem

Thermelgy-Repo, 2022-03-24:
Dear team,

Thank you for your this wonderful library, 

I am getting a strange issue, I am trying to build a non-blocking GSM library in ESP32 using espsoftserial. My function is perfectly working without FreeRTOS. But when I put the function in a FreeRTOS task with same priority, am getting garbage value while recieving (read() API). If I changed the task priority higher, it works fine. But in  my case I want to use the two task with same priority. The similar issue is found on #149, He solved it by changing the priority, In my case I cannot change the priority. I found your latest release V6.15.2 that it reduced the ISR and FreeRTOS is directly supported. Kindly guide me to solve this issue.

```
byte QM66::sendATcommand2(char* ATcommand, char* expected_answer1, char* expected_answer2, unsigned int TimeOut, unsigned int repetitions) // AT command, Correct answer, Error message, TimeOut
{

  unsigned x = 0, cnt = 0;
  byte answer = G_NULL;

  unsigned long Prev=0;

  while (cnt < repetitions && answer != G_OK) 
  {
    byte answer = G_NULL;

    memset(response, '\0', Res_Size);    // Initialize the response string

  //delay(100);
  //Serial2.flush();
  
    Serial.print(F("\nCommand : "));    Serial.print(ATcommand);    // Send the AT command
    
    SerialAT.println(ATcommand);    // Send the AT command

    x = 0;
    Prev = millis();

    // this loop waits for the answer
    do 
    {
      // if there are data in the UART input buffer, reads it and checks for the asnwer
      if (SerialAT.available()) 
      {
        if(x < Res_Size - 1) // '/0' should also consider
        {
          response[x++] = SerialAT.read();
        }

          
        // check if the desired answer 1  is in the response of the module
        if(strstr(response, expected_answer1) != NULL)
        {
          answer = G_OK;
          break;
        }
        // check if the desired answer 2 is in the response of the module
        else if(strstr(response, expected_answer2) != NULL)
        {
          answer = G_NOTOK;
        }

        else
        {
          
        }

      }

    } while(((millis() - Prev) < TimeOut) && (answer == G_NULL));
/*
    for(int i=0; i<50; i++)
    {
      Serial.print(response[i]);
    }
*/
    cnt++;
    Serial.print("\nt="); Serial.print(millis() - Prev);

    if(answer == G_OK)
    {
      Serial.print(F(", Sucess Resp : ")); Serial.print(response);
      vTaskDelay(100 / portTICK_PERIOD_MS); // Delay of 100ms in each sequence
      break;
    }

    else if((answer == G_NOTOK) )
    {
      Serial.print(F(", Error Resp : ")); 
    }
    
    else if(answer == G_NULL)
    {
        Serial.print(F(", Resp Timeout : "));
        answer = G_TIMEOUT;
    }

    else
    {

    }

    Serial.print(response);

    vTaskDelay(100 / portTICK_PERIOD_MS); // Delay of 100ms in each sequence
  
 }
  
  return answer;
}

```

dok-net, 2022-03-26:
I don't know that FreeRTOS is directly supported by this library - in fact, it's for the ESP8266 and working on the ESP32 is a welcome bonus, but given the 3 HW UARTs there not a primary focus; the maintainers of the ESP32 Arduino project are the first to tell you the same.

As for receiving, there are two areas that come to my mind: core-affinity and buffer sizes. Please educate yourself on the buffer sizes EspSoftwareSerial allows you to configure. And check your core affinity with regard to IRQ handling in your FreeRTOS BSP.

dok-net, 2022-04-09:
I will be closing this shortly?

Thermelgy-Repo, 2022-04-11:
> I don't know that FreeRTOS is directly supported by this library - in fact, it's for the ESP8266 and working on the ESP32 is a welcome bonus, but given the 3 HW UARTs there not a primary focus; the [...]

Dear @dok-net , Thank you so much for your valuable suggestion, I temporally resolved it by using the SoftSerial in the loop task without creating a separate task. But I am facing a new challenge, the same issue is happening while using with Hardware serial function. I am expecting it to be problem of IRQ. It will be a great help, if you can help on this.

dok-net, 2022-04-11:
It's really too difficult figuring out what you are reporting, when your sketch is quite complex - even with commented code in there. Please remember the MCVE rule. Also, switching in the same issue from one problem to another, doesn't help. If this one is resolved, close it. Then open another one, with said MCVE and exact details pertaining to one, and only one, issue at a time.
I am not promising I will be available to help, but going this path at least keeps the issue tracking clean, and someone else might be able to point you in the right direction.

closed 2022-05-05 by dok-net