plerup/espsoftwareserial #234: section `.text1' will not fit in region `iram1_0_seg' on ESP8266

ali80, 2022-04-03:
Hi, I'm getting this linking error on ESP8266, when I remove `IRAM_ATTR`, the project links without any problem, Is there a way to fix this properly?

dok-net, 2022-04-03:
You probably are using a separate version of EspSoftwareSerial for some reason, which you are not supposed to do. Matching releases are part of the Arduino Core for ESP8266 distribution, please stick with those.

ali80, 2022-04-04:
I'm using PlatformIO
```bash
 - framework-arduinoespressif8266 3.30002.0 (3.0.2) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-xtensa 2.100300.210717 (10.3.0)
```
first I tried using the builtin softwareserial but got the exact same error, thought maybe the builtin version has a bug, 
then tried the package in platformio registery which is currently `plerup/EspSoftwareSerial@^6.15.2`, got the same error,
then tried getting the package directly from git, same error again.

ali80, 2022-04-04:
this is a fairly large project, it seems that I'm running out of IRAM! 
disabling some other modules fixes the problem, 
I think the main use case of `espsoftwareserial` would be large projects, is there a way to reduce IRAM usage or increase IRAM value on ESP8266?

ali80, 2022-04-04:
If you wanna increase IRAM space, this is the way
[Adjusting ICACHE to IRAM ratio](https://docs.platformio.org/en/latest/platforms/espressif8266.html#id22)

closed 2022-04-04 by ali80