plerup/espsoftwareserial #236: Future Request OneWire

bulentperktas, 2022-04-24:
is it possible to add another config that open-drain output while sending transmit(TX) data? Your onewire implementation is amazing, i think that float driven TX output give a chance that long distance communication. Simply added a 4k7 (or convinent value) pullup resistor on the bus same as onewire driven sensors.

any idea?

Thank you
Bulent

dok-net, 2022-04-25:
There have been different ways of driving the output signal (output; open collector; toggling input w/ pullup and open collector output) over the course of the last two years; the ways it is now seems to fit most/all users. You see to be asking from a purely theoretic point of view. For driving long distance lines, EIA232 isn't the most appropriate hw protocol selection to begin with. If you could show that, for instance, an EIA 485 line driver requires open collector output, that would a real reason to implement that. Otherwise, you might prepare a PR and we will discuss that.

bulentperktas, 2022-04-25:
Dear Kaar @dok-net Thank you for your reply,

I found that all your explanation for pin direction modes such as output; open collector; toggling input w/ pullup and open collector choices available both on ESP32 and ESP8266 Arduino core,

for ESP32 [esp32-hal-gpio.h](https://github.com/espressif/arduino-esp32/blob/b3c203db26f87efd965f6c3f6cc078f1b3f2c5db/cores/esp32/esp32-hal-gpio.h#L43) and [esp32-hal-gpio.c](https://github.com/espressif/arduino-esp32/blob/b3c203db26f87efd965f6c3f6cc078f1b3f2c5db/cores/esp32/esp32-hal-gpio.c#L92) library

```
//GPIO FUNCTIONS
#define INPUT             0x01
// Changed OUTPUT from 0x02 to behave the same as Arduino pinMode(pin,OUTPUT) 
// where you can read the state of pin even when it is set as OUTPUT
#define OUTPUT            0x03 
#define PULLUP            0x04
#define INPUT_PULLUP      0x05
#define PULLDOWN          0x08
#define INPUT_PULLDOWN    0x09
#define OPEN_DRAIN        0x10
#define OUTPUT_OPEN_DRAIN 0x12
#define ANALOG            0xC0
```
for ESP8266 [Arduino.h](https://github.com/esp8266/Arduino/blob/f60defc3d306152c13a20065aaa3d81ea17e690b/cores/esp8266/Arduino.h#L47) and [core_esp8266_wiring_digital.cpp](https://github.com/esp8266/Arduino/blob/da6ec83b5fdbd5b02f04cf143dcf8e158a8cfd36/cores/esp8266/core_esp8266_wiring_digital.cpp#L50) library

```
//GPIO FUNCTIONS
#define INPUT             0x00
#define INPUT_PULLUP      0x02
#define INPUT_PULLDOWN_16 0x04 // PULLDOWN only possible for pin16
#define OUTPUT            0x01
#define OUTPUT_OPEN_DRAIN 0x03
#define WAKEUP_PULLUP     0x05
#define WAKEUP_PULLDOWN   0x07
#define SPECIAL           0xF8 //defaults to the usable BUSes uart0rx/tx uart1tx and hspi
#define FUNCTION_0        0x08
#define FUNCTION_1        0x18
#define FUNCTION_2        0x28
#define FUNCTION_3        0x38
#define FUNCTION_4        0x48
```

Advantage of open-collector outputs, or open-drain outputs is that the load to be switched or controlled can be connected to a voltage supply which is independant, and/or different from the supply voltage used by the controlling circuit, and that they can “sink” or “source” an externally-supplied voltage depending upon whether its to ground, or source.

Another advantage is that more than one open-collector output can be connected to a single line. If all outputs attached to the line are in the high-impedance state, the pull-up resistor will hold the wire in a high voltage (logic 1) state. If one or more device outputs are in the logic 0 (ground) state, they will sink current and pull the line voltage toward ground. 

Thank you,
Bulent

dok-net, 2022-05-28:
@bulentperktas Would you like to test and report back on #239 ?

dok-net, 2022-07-10:
@bulentperktas 1.5 months and still no word. I will close this on short notice.

dok-net, 2022-07-23:
@bulentperktas Another friendly, final reminder; I would like to merge my PR, but it should be tested. You suggested the change, I was hoping you could try it and give feedback. If not, I am not sure if it will make it anytime soon. Which would be a loss to you as much as anybody else. RSVP.

dok-net, 2022-12-04:
Fixed by #239

closed 2022-12-04 by dok-net